Ultimaker/Cura #19222: Slicing for Dremel 3D45 fails

daUser, 2024-06-11:
### Cura Version

5.7.2

### Operating System

linuxserver/cura:latest

### Printer

Dremel 3D45

### Name abnormal settings

Cooling => Minimum Speed is changed to 1.00mm/s

### Describe your model

Slices in Cura 5.7.1 under Windows, thingsverse.com/thing:4694600

### Add your .zip here ⬇️

[Stackable Calibration cube - 4694600.zip](https://github.com/user-attachments/files/15794011/Stackable.Calibration.cube.-.4694600.zip)

GregValiant, 2024-06-11:
Thanks for the report.
The model file doesn't do us any good.  We need a project file and the Cura log file to investigate.  The instructions are on the page you filled out for this report.

The model itself is fine and I didn't have any trouble slicing it.  Given just the model I can't duplicate the problem.
[image]

daUser, 2024-06-12:
oh, excuse me, was a long day for me.

[D_XYZ_cal_cube_LEGO_v3.zip](https://github.com/user-attachments/files/15803514/D_XYZ_cal_cube_LEGO_v3.zip)
i found the logs at /config/.local/share/cura/5.7 - is this correct or are there other log-files you need?
[cura.log](https://github.com/user-attachments/files/15803530/cura.log)

GregValiant, 2024-06-12:
The Dremel printers don't have definition files in Cura so the two plugins have to cover that.
The log file indicates errors in regards to the plugins.  That might be the result of the plugins not being updated to the current Cura version.
An odd thing was that the printer installed with no "firmware flavor".  It should have defaulted to Marlin.

After I installed the Dremel Printer plugin the slice went well without any problem.  I was able to generate a "g3drem" file as well.
I'll leave the bug label on this but you may have to contact the authors of the Dremel plugins and see if there have been other issues.  One of them has a Github site [HERE](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/issues).
You might also check on Reddit and see if others are having problems.

daUser, 2024-06-13:
Thank you very much.

I have opened a issue, an after that i think i found the "Problem": the actual Version of the Plug in is 1.0.0 - installed by the Cura Marktplace is 0.8.1.

Is there a way for a manual installation?

GregValiant, 2024-06-13:
Maybe.
Use the "Help | Show Configuration Folder" command in Cura.  Within the config folder should be a folder "plugins" and one of the plugins should be the "DremelPrinterPlugin" folder.
Simply downloading the new version of the plugin, unzipping it, and pasting it into the Plugins folder (and over-writing the older version) may work.  Cura would need to be restarted to acknowledge the change.
I didn't install the second Dremel plugin regarding network printing as I didn't think it would work with my "virtual" 3D45 anyway.

GregValiant, 2024-06-21:
This problem appears to be in the "Quality" files that are included with the Dremel plugin.
This is example is from "Dremel_3D45_dremel_eco_abs_draft.inst.cfg" and all 30 quality files are the same.
**cool_min_speed = 0**

But from the base fdmprinter.def.json file:
"cool_min_speed":
                {
                    "label": "Minimum Speed",
                    "default_value": 10,
                    **"minimum_value": "1","**

So your quality files are setting the Minimum Speed to "0" which is an illegal value.
This is not a Cura problem but rather something that is occurring with 3rd party software (the definitions).  Hopefully the author can fix this for you.  You could go into each quality file and set each minimum speed to 10.  That would fix the problem.

I'm going to remove the bug label and close this as it doesn't appear to be a Cura issue.  Let me know of you think that is premature.

closed 2024-06-21 by GregValiant