Ultimaker/Cura #19223: Model looks different when viewed in Preview vs Prepare

buteomont, 2024-06-12:
### Cura Version

5.7.0

### Operating System

Ubuntu 20.04

### Printer

Tronxy

### Reproduction steps

1. Load the model.  Looks fine in Prepare tab
2. Switch to the Preview tab. Model has large top hole plugged.

### Actual results

Model has large top hole plugged when viewing in Preview tab. Hole is open as it should be in Prepare tab.

### Expected results

Hole should be open in Preview tab as well.

### Add your .zip and screenshots here ⬇️

[faucetHead-Cura.zip](https://github.com/user-attachments/files/15796656/faucetHead-Cura.zip)
[image]
[image]

buteomont, 2024-06-12:
Note: If I remove the fillets on the edges of the model the problem goes away.

Asterchades, 2024-06-12:
The Prepare screen only shows a raw preview of the geometry of any model you load. Almost nothing about the surfaces is taken into account - everything, including the lighting shown, is handled strictly by the coordinates of what it loads. There is no additional processing done beyond this.

When you slice the actual structure of the model is taken into consideration - the direction each polygon is "facing," and what they are (or aren't) attached to. When you import this model into Cura it informs you (quite rightly) that it is not manifold, and indicates three detached surfaces with a pink/teal polkadot pattern. The three surfaces are the outward fillet at the bottom of the fitting, the inward "fillet" at the top of the bulb, and the outside wall of the fitting (from the inside only).

Closer inspection of this model shows that the threads in the fitting do not have the crest filled in, leaving it completely open, which is both causing the outside of the fitting to show as detached and is likely responsible for the fitting itself being filled in. The visible burgundy is the inside of the outer-most wall, which should not be visible through the threads (and correctly isn't right down near the bottom).
[image]

The external "join" between the fitting and bulb also don't come into contact with each other, regardless of how close they are, causing the external surface to be non-continuous; though because of the overlap between them Cura can probably fill in the gap. Again the burgundy visible is the inside of the upper part, which should not be visible from the outside.
[image]

I would suspect that whatever program you're using is simply failing to export the thread geometry correctly when you're adding that fillet process. I could only begin to guess why. I also suspect that the bulb is too small given the offset in the outer wall and pronounced ledge in the flow path, however that could be a deliberate decision by you.

Because of how "obvious" the issues are from a modelling point, MS' 3D Builder can repair this model to the point it will slice successfully by stitching the missing crest vertically and the broken external seam horizontally, though it did take a couple of minutes to do it even on my 5800X3D.

For more information on why non-manifold geometry causes issues with slicing, I would encourage you to read the following article:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/missing_parts.md#non-manifold-meshes

closed 2024-06-12 by Asterchades

buteomont, 2024-06-12:
Thank you for the excellent explanation. I should have examined the model more closely; the problem is obvious now that it's been explained. I was not aware that the prepare and preview views were handled so differently.

Perhaps something a little more obvious, such as a pop-up or error message, when a non-manifold model is loaded would help.  But not being familiar with the internals of the code, I don't know how difficult it would be to add such a feature.

Asterchades, 2024-06-12:
Grab the **Mesh Tools** plugin from the marketplace. This will present an error if non-manifold geometry is present, and also has some very rudimentary tools to try and correct some issues like inverted normals.

I actually thought the error was built-in functionality, but I guess I've had that plugin installed for a very long time.