Ultimaker/Cura #19228: Remove the hotend start temp, or put it in the slicer Startup g-code section.

GregGFlynn, 2024-06-13:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

Start G-code setting in Cura is set to this:

; Ender 3 Custom Start G-code
M413 S0 ; DPLR
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 P1 ; Home automatically and run mesh leveling on every print
G29 A ; Activate UBL
C108 ; Close the mesh viewer (optional)
G1 Z5.0 F3000 ; Move Z Axis up a bit during heating to not damage bed
M104 S{material_standby_temperature} ; Start heating up the nozzle most of the way
M190 S{material_bed_temperature_layer_0} ; Start heating the bed, wait until target temperature reached
M109 S{material_print_temperature_layer_0} ; Finish heating the nozzle
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

### Actual results

G-Code saved in file looks like this

M104 S215
M105
M109 S215
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M413 S0 ; DPLR
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 P1 ; Home automatically and run mesh leveling on every print
G29 A ; Activate UBL
C108 ; Close the mesh viewer (optional)
G1 Z5.0 F3000 ; Move Z Axis up a bit during heating to not damage bed
M104 S175 ; Start heating up the nozzle most of the way
M190 S60 ; Start heating the bed, wait until target temperature reached
M109 S215 ; Finish heating the nozzle
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

### Expected results

M104 S215
M105          Why are you heating up the hotend now, only to cool it down and back up a few commands later?
M109 S215

M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M413 S0 ; DPLR
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 P1 ; Home automatically and run mesh leveling on every print
G29 A ; Activate UBL
C108 ; Close the mesh viewer (optional)
G1 Z5.0 F3000 ; Move Z Axis up a bit during heating to not damage bed
M104 S175 ; Start heating up the nozzle most of the way
M190 S60 ; Start heating the bed, wait until target temperature reached
M109 S215 ; Finish heating the nozzle

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-06-13:
Thanks for the report.
This is a duplicate of #19204 .  The Cura team is aware of the issue.  The prepend code is there in case there are no temperature "keywords" in the StartUp Gcode.  It was broken in 5.7.2 by some other change that was made.

This post processor addresses the problem and removes the offending lines.  Unzip the file and place "CuraPrependBugfix" into your Configuration folder ("Help | Show Configuration Folder") and the "scripts" sub-folder.
Use the "Extensions / Post Processing \ Modify Gcode" command then then "Add Script".
Make sure the post processor "enabled" checkbox is checked.
[CuraPrependBugFix.zip](https://github.com/user-attachments/files/15823167/CuraPrependBugFix.zip)
I'll go ahead and mark this as a duplicate and close it.  I hope you understand.

closed 2024-06-13 by GregValiant

GregGFlynn, 2024-06-13:
Greg,

   Thanks for the fast reply!  I usually look at the G-code file in a text
editor jst to make sure all my startup codes are correct, so for now I'll
just delete those lines manually rather than using a post processing code.

Thanks,

Greg