Ultimaker/Cura #19229: On mac, when you open fullscreen, dialogs open on another desktop, somewhere

christhomas, 2024-06-13:
### Cura Version

5.7.1

### Operating System

Mac Os Catalina

### Printer

Ender 3 s1 pro

### Reproduction steps

When you open any setting or preference dialog, in fullscreen mode on a mac, the dialog box will appear on another desktop somewhere, instead of over the top of the application

Perhaps this is a problem of the fullscreen mode on mac. But if this is the case. A method should be found where the dialog boxes open inside the app window, or fix the issue where it's on another screen somewhere. 

### Actual results

The dialog pops up somewhere, but I have no idea where, I need to go hunting through my screens and desktops until I find it. 

### Expected results

The dialog should either appear in the window, or overlaid on the window itself. But not hidden on another desktop somewhere that I have to go hunting for it

### Add your .zip and screenshots here ⬇️

I have no zip or screenshot that would help

The--Captain, 2024-06-14:
Not surprised.

https://github.com/Ultimaker/Cura/issues/14076

christhomas, 2024-06-14:
I love how the first reply on that ticket basically suggests the problem is that he has too many things installed instead of answering the critique that basic dimensions on user interface elements are not installed. I mean, he could have 100 extruders installed. The problem is still that the user interface have no control over dimensions.