Ultimaker/Cura #19232: Error in all projects

ciberdraco, 2024-06-14:
### Cura Version

5.7.1

### Operating System

windows 11

### Printer

artillery x2

### Describe your model

I have problem with all models these last two weeks

### Add your .zip here ⬇️

[ASX1_Flexi Town Stormtrooper.zip](https://github.com/user-attachments/files/15842314/ASX1_Flexi.Town.Stormtrooper.zip)

I have problem with all    models. I've tried other cura versions for example 5.6. 

I think that the problem is the compoter but i dont know why. if I use old computer with the same profile, Cura works correctly, but in this new computer its not working. Maybe exist a log in order to try to found the problem.

thank you

GregValiant, 2024-06-14:
Thanks for the report.
In Cura use the "Help | Show Configuration Folder" command.  In that folder is "Cura.log".  Zip the file and post the zip folder here.  
The Cura team will need the log file to troubleshoot this.
You can try right-clicking on the Cura desktop icon and selecting "Run as Administrator".  Sometimes that works.

ciberdraco, 2024-06-15:
hi team, this is the log file.
[cura.log](https://github.com/user-attachments/files/15845797/cura.log)
thnks you