Ultimaker/Cura #19235: Crashes on first run when skipping sign in or signing in

masoncfrancis, 2024-06-14:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

N/A

### Reproduction steps

- Start Cura after freshly installing it. 
- Click through the screens until you reach the one where you are prompted to sign in or skip
- Skip or sign in

### Actual results

Upon clicking skip, Cura pauses and then closes completely without showing any error message. Upon clicking sign in, I am given the opportunity to sign in using my browser but after that's done, Cura pauses and then completely similar to when I click skip. 

### Expected results

Cura should continue in the setup process. 

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/15844685/cura.log)

burakturnaoglu, 2024-07-01:
Its worked for me

https://github.com/Ultimaker/Cura/issues/19198#issuecomment-2199438892

TheVeridian, 2024-08-05:
Bumping this issue, as I'm experiencing the same issue on a brand new laptop that had a clean install of Windows 11. Attempted the solution provided with the same result.

@masoncfrancis did you ever solve your issue? Thx.

masoncfrancis, 2024-08-05:
@TheVeridian I downgraded to 5.6.0 and that version worked, so I’ve just been using that until something gets figured out. 

Interestingly, yesterday after using 5.6.0 for over a month, I tried installing 5.8.0 alongside 5.6.0 and it started working fine 🤷‍♂️

JustTheBast, 2024-12-14:
I believe this is an issue with the _next_ step in the "first time onboarding" process, which is to select your printer.  Whether you "skip" or actually log in, in either case, the program next attempts to display the "add printer" dialogue and _that_ causes the crash.  The issue I found relating to that topic is #19924 

As I commented on that issue, the last version that could still open the "add printer" dialogue (and thus also proceed through onboarding) was version 5.6.0.  As a workaround, one can install 5.6.0 in parallel to the latest version, go through first time setup with that, and then switch to the latest version for the actual work.  Just remember to use the old version, if you ever need to add another printer.

Kilian654, 2025-01-11:
Same issue on Linux Mint with AppImages. Last working version is 5.6.0. The account sign in, create or skip closes cura everytime.