Ultimaker/Cura #19238: 5.7.2 Files sliced for klipper running in the wrong order

MikeOxawopper, 2024-06-15:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Klipper Ender3 v2

### Reproduction steps

1. import object
2. slice

### Actual results

When the print starts the order of operation is all messed up

1. it heats the nozzle to initial layer temp
2. heats the bed to initial layer temp
3. homes the printer
4. parks and raises the tool head
5. cools the nozzle to 120
6. meshes the bed via KAMP

i stopped it at this point

### Expected results

It should and does in every other version of Cura 5 x x

1. home the printer
2. heat the bed to initial
3. park and raise the tool head
4. heat the tool head to 120
5. mesh the bed with KAMP
6. park and raise the tool head
7. heat nozzle to inital temp
8. print the object

### Add your .zip and screenshots here ⬇️

[Pinder Mount New.zip](https://github.com/user-attachments/files/15847751/Pinder.Mount.New.zip)

GregValiant, 2024-06-15:
Thanks for the report.
This sounds like a duplicate of #19204 .  There is some discussion there and I've posted a post-processor to fix the problem described there.  Let me know if it sounds like the same bug.
If you open a gcode file in a text editor - do you see these lines:
;Generated with Cura_SteamEngine 5.7.2
**M104 S210
M105
M109 S210**
M82                            ;absolute extrusion mode
The M104 and M109 lines should not have been added because I have the build plate temperature and the hot end temperature called out in my StartUp gcode using Cura keywords.  That's the bug.

MikeOxawopper, 2024-06-16:
i added the Prepend script to Cura 5.7.2 from the post #19204 and it seems to be working normally again

GregValiant, 2024-06-17:
Since this is sorted out for now I'll go ahead and close it in favor of the first report.  I hope you understand.  If you have any thoughts or comments you can post them on the other page.

closed 2024-06-17 by GregValiant