Ultimaker/Cura #19241: Plugins

de-vel-op-r, 2024-06-16:
### Cura Version

4.13.0

### Operating System

Ubuntu 22.04

### Printer

Creality CR-10S

### Reproduction steps

1. Install Cura for first time on clean machine. Install Cura from Ubuntu Software.
2. Started Cura.

### Actual results

1. After starting Cura, 2 plugins produce messages that they may need installation to fix problem. In Marketplace, Community I selected AMF Reader and Trimes Reader. Both were unselected. Quit Cura for install to take place, which was instructions on screen. Restarted Cura. Same problem. 
2. Uninstalled Cura. Installed Cura. Repeat of step 1 above. Same problem.
3. Selected to Uninstall both plugins from the warning messages. Restart Cura. Same problem.

### Expected results

Cura to install without issues like this.
Send me a message when this is fixed. I need this ASAP.
[image]

### Add your .zip and screenshots here ⬇️

N/A

de-vel-op-r, 2024-06-16:
You can dismiss this issue. I have located and resolved the issue. Depreciated import imp module in the plugin library.

closed 2024-06-16 by de-vel-op-r

fieldOfView, 2024-06-17:
FYI, the problem is most likely that whoever packaged Cura for the Ubuntu store omitted a required dependency: 
https://pypi.org/project/trimesh/

Ultimaker only makes (and supports) an appimage for Linux. ALL other distributions are packaged by third parties, and they may miss certain dependencies.