Ultimaker/Cura #19242: [5.4.0 and 5.7.2] Part section not attached to rest of part even though model is not designed that way

itzameongithub, 2024-06-17:
### Cura Version

5.4.0, 5.7.2

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro, Creality CR-10S, Monoprice Maker Select Plus

### Reproduction steps

Issue occurred with 5.4.0. Upgraded to 5.7.2 and it still encounters the same problem. Attempted to slice with multiple different  printer models, layer thicknesses, and wall/infill settings.

### Actual results

The circle/triangle of the part become separate parts of the print. The problem is that the design was formed by indenting the design into a solid wall. They are not, in fact, separate parts of the model to be attached. They should be a part of the wall structure. However, when Cura attempts to slice it, the circle and triangle actually become separate print areas with their own walls instead of being incorporated into the main structure. The external circle, however, becomes a part of the main body as expected. Because of this, parts of that triangle have cleanly broken off during sanding. This would not happen if they were incorporated into the body as expected. The indentation only goes about 1.5mm into a part with a thinkness of 4.9mm.

### Expected results

The circle and triangle design should be incorportated into the part as a whole with the walls extending out into the circle and triangle design, just like is done with the outside ring.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[Side Wall A v5.zip](https://github.com/user-attachments/files/15860417/Side.Wall.A.v5.zip)

itzameongithub, 2024-06-17:
Additional screenshot showing how the center circle and triangle segments are sliced as being separate from the main body when they're supposed to be incorporated into the part. The external ring is being incorporated as expected. Again, the circle and triangle are not separate model objects but are formed by indenting the outline of the design into a solid wall. They should be a part of the structure just like the outside ring is.
[image]

itzameongithub, 2024-06-17:
Close issue. After further examination, the problem has been isolated to issues in the CAD software, not the slicer.

GregValiant, 2024-06-17:
I knew that.
I'll go ahead and close this.  Good luck with it.

closed 2024-06-17 by GregValiant