Ultimaker/Cura #19243: Loading STL with space in name fails

raphaelsiz, 2024-06-17:
### Cura Version

5.7.2

### Operating System

Debian Linux

### Printer

Anycubic i3 Mega S/Pro

### Reproduction steps

1. Open cura from terminal specifying stl file with spaces in its name, `./Ultimaker-etc Button\ etc.stl`

### Actual results

Cura opens with nothing on the build plate and an error message at the bottom of the screen that it couldn't open the file.

### Expected results

Cura should open with the stl on the build plate.

### Add your .zip and screenshots here ⬇️

Log messages when I tried to open a file with a space in the name:
```
2024-06-16 23:21:04,449 - INFO - [MainThread] cura.CuraApplication.readLocalFile [2006]: Attempting to read file file:///home/morgan/Button hole drill gears.stl
2024-06-16 23:21:04,459 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]: Exception: Reading stl file failed.
2024-06-16 23:21:04,465 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]: Traceback (most recent call last):
2024-06-16 23:21:04,466 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]:   File "/tmp/.mount_UltiMa73jrtB/share/uranium/plugins/FileHandlers/STLReader/STLReader.py", line 32, in load_file
2024-06-16 23:21:04,466 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]:     self._loadWithNumpySTL(file_name, mesh_builder)
2024-06-16 23:21:04,466 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]:   File "/tmp/.mount_UltiMa73jrtB/share/uranium/plugins/FileHandlers/STLReader/STLReader.py", line 76, in _loadWithNumpySTL
2024-06-16 23:21:04,467 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]:     for loaded_data in stl.mesh.Mesh.from_multi_file(file_name, mode=stl.stl.Mode.AUTOMATIC):
2024-06-16 23:21:04,467 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]:   File "stl/stl.py", line 344, in from_multi_file
2024-06-16 23:21:04,467 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]:     fh = open(filename, 'rb')
2024-06-16 23:21:04,467 - ERROR - [JobQueueWorker [2]] STLReader.STLReader.load_file [34]: FileNotFoundError: [Errno 2] No such file or directory: '/home/morgan/Button hole drill gears.stl'
2024-06-16 23:21:04,472 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]: Exception: Exception occurred while loading file /home/morgan/Button hole drill gears.stl
2024-06-16 23:21:04,475 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]: Traceback (most recent call last):
2024-06-16 23:21:04,475 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:   File "UM/FileHandler/ReadFileJob.py", line 82, in run
2024-06-16 23:21:04,475 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:     self.setResult(self._handler.readerRead(reader, self._filename))
2024-06-16 23:21:04,476 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:   File "UM/Mesh/MeshFileHandler.py", line 36, in readerRead
2024-06-16 23:21:04,476 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:     results = reader.read(file_name)
2024-06-16 23:21:04,476 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:   File "UM/Mesh/MeshReader.py", line 26, in read
2024-06-16 23:21:04,476 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:     result = self._read(file_name)
2024-06-16 23:21:04,476 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:   File "/tmp/.mount_UltiMa73jrtB/share/uranium/plugins/FileHandlers/STLReader/STLReader.py", line 52, in _read
2024-06-16 23:21:04,476 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]:     if numpy.amin(verts[:, 1]) == numpy.amax(verts[:, 1]):
2024-06-16 23:21:04,477 - ERROR - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [84]: TypeError: 'NoneType' object is not subscriptable
```

Log messages when I did the same thing after changing the name of the file not to have spaces in it:
```
2024-06-16 23:29:03,443 - INFO - [MainThread] cura.CuraApplication.readLocalFile [2006]: Attempting to read file file:///home/morgan/Downloads/button_gears.stl
2024-06-16 23:29:03,454 - DEBUG - [JobQueueWorker [2]] UM.Mesh.MeshData.calculateNormalsFromVertices [561]: Calculating normals took 0.0006153583526611328 seconds
2024-06-16 23:29:03,456 - DEBUG - [JobQueueWorker [2]] STLReader.STLReader._read [68]: Loaded a mesh with 10416 vertices
2024-06-16 23:29:03,466 - DEBUG - [JobQueueWorker [2]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0025937557220458984 seconds. 117 input vertices. 117 output vertices.
2024-06-16 23:29:03,474 - DEBUG - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [87]: Loading file took 0.0 seconds
```
[Full cura.log](https://github.com/user-attachments/files/15862841/cura.log)

Originally thought it was what's happening [in this issue](https://github.com/Ultimaker/Cura/issues/18892) until disabling USB printing did nothing. But I saw the weird formatting of the file name in the error and tried removing spaces from the file name, and that seemed to have fixed it (or coincidentally it started working again!)

GregValiant, 2024-06-17:
Thanks for the report.
Have you tried quotation marks around the file name string?

smartavionics, 2024-06-17:
Surely that should be ./Ultimaker-etc\ Button\ etc.stl  ?

raphaelsiz, 2024-06-28:
@GregValiant I apologize for not checking that before making the issue, but I checked just now and the same command with quotes around the file name (see below) also didn't work.
```
./Ultimaker-etc "button gears.stl"
```
Had the same effect as before. But interestingly, when I checked what I thought fixed it before (no space in the name), that didn't fix it either.

But then I looked closer at the error message and saw that it said there was no file at `/home/user/button_gears.stl`, I had run the program from `/home/user/Downloads`, so I had thought `button_gears.stl` as an argument would lead the computer to `/home/user/Downloads/button_gears.stl`.

Surely enough, when I cd'ed into my home directory, the app worked just fine, including with spaces in the file name. So I seem to have completely misunderstood the problem. I guess before the actual change I made (that I somehow didn't notice) was that **Cura only worked if I ran it from my home directory**, because it seems to assume that I am when parsing the file input.

I'm not sure what the etiquette is here. I probably shouldn't have tried to pinpoint the source in the title of the issue. It's still an issue, but apparently it wasn't the one I thought it was! Sorry about that.

As far as the comment from @smartavionics, No, the `./Ultimaker-etc` is the program and the `Button\ etc.stl` is the file I'm trying to open in it. There should be an unescaped space between the two.

raphaelsiz, 2024-06-28:
For the record, as additional information, I did attempt putting a `./` before the file name because some programs are strict about that, but that didn't fix it, only changing directory did (and once I changed directory, it didn't mind that I said `Downloads/button\ gears.stl` instead of `./Downloads/button\ gears.stl`)

smartavionics, 2024-06-28:
>As far as the comment from @smartavionics, No, the ./Ultimaker-etc is the program and the Button\ etc.stl is the file I'm trying to open in it. There should be an unescaped space between the two.

My mistake, sorry.