Ultimaker/Cura #19250: Slicing failed with unexpected error

iuliupop, 2024-06-18:
### Cura Version

5.7.0 beta1

### Operating System

Windows 10

### Printer

Creality Ender 5 S1

### Reproduction steps

Imported SingleObject-FasterPrint.STL with current slicing settings (see attached settings).

[Ender 5 S1 Printer settings in Cura.zip](https://github.com/user-attachments/files/15888338/Ender.5.S1.Printer.settings.in.Cura.zip)
[SingleObject-FasterPrint.zip](https://github.com/user-attachments/files/15888339/SingleObject-FasterPrint.zip)
[Cura_settings.zip](https://github.com/user-attachments/files/15888344/Cura_settings.zip)

### Actual results

Slicing started and after several seconds slicing stops with error.

### Expected results

Slicing complete without errors:
Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker.

### Add your .zip and screenshots here ⬇️

[cura_5.7.0_cura.zip](https://github.com/user-attachments/files/15890250/cura_5.7.0_cura.zip)

GregValiant, 2024-06-18:
Thanks for the report.
There doesn't appear to be any errors in the model and it sliced for me with my Ender 3 Pro active and the standard profile with normal supports and PLA.
You are still using the 5.7.0beta version.  That is a test version before the release of 5.7.0 stable.  The current Cura version is 5.7.2.  Please update to a stable version be it 5.7.0, 5.7.1, or 5.7.2.  Be aware that 5.7.2 has a bug that affects the startup.  If you decide up upgrade to that one there is a post processor posted in the bug report (#19204 ) that will fix it.

From your log file is a generic error:  "Backend exited abnormally with return code 3221225477!"  The cura team should have your crash report.  

I installed an Ender 5 S1 and your curaprofile and material and the slice finished but it took a while.
"Slicing took 1141.870280265808 seconds".  It's nice that it goes to 12 decimal places.  Rounding it off it comes out to 19 minutes to slice.  (For comparison - the slice using my printer and normal supports took 4 minutes.)

This is in 5.7.2.
[image]

I had to rebuild the project from the files you supplied.  If you would load the model into Cura and set up your slice, and then use the "File | Save Project" command and zip and post that 3mf file things would be much easier.  It will contain the model, your printer, and all your settings.

Typically I would close this because it is for a _'2 version out of data'_ beta, but I'll leave it open for the Cura team as the slicing time seemed really long for a model with only 140k surfaces.

iuliupop, 2024-07-02:
With cura default settings it slices ok, but I have several custom settings that I need and with those I get these errors. I will get back with the 3mf file attached too.

iuliupop, 2024-07-02:
Please find attached the 3mf file also.
[CE5S1_SingleObject-FasterPrint_3mf.zip](https://github.com/user-attachments/files/16061761/CE5S1_SingleObject-FasterPrint_3mf.zip)

iuliupop, 2024-07-02:
I will try slicing with 5.7.2 and get back with details.

GregValiant, 2024-07-02:
I'm not having any trouble with it.
My computer must have been having a bad day.  The slice took about 4 minutes.
Just for reference, I print TPU on my bowden printer at 12mm/sec.  120mm/sec seems excessive for TPU, but you know your printer.

iuliupop, 2024-07-05:
I confirm that with version 5.7.2 the slicing was successful and it took around 8 minutes to complete.

iuliupop, 2024-07-05:
> I'm not having any trouble with it. My computer must have been having a bad day. The slice took about 4 minutes. Just for reference, I print TPU on my bowden printer at 12mm/sec. 120mm/sec seems excessive for TPU, but you know your printer.

Hi GredValiant,
printing this TPU (from eSUN 95A transparent red) on the Ender 5 S1 at 120mm/sec prints very good (same quality as for 20 or 40 mm/sec). I did not try higher speeds as I red that TPU needs to be printed at lower speeds. I still need to fine tune the retraction and support settings (as I still have some fine strings when printing and the supports are harder to remove).
[image]
[image]