Ultimaker/Cura #19256: Missing printers

LiesRom, 2024-06-19:
### Is your feature request related to a problem?

There are some printers missing from the printer-overview (such as flashforge guider 2s and creality ender 5 pro).

### Describe the solution you'd like

Is it possible to add these printers to the list?

### Describe alternatives you've considered

Install a different software

### Affected users and/or printers

I really like your software. It is very useful that you have adapted it to many types of printers.

GregValiant, 2024-06-19:
All of the non-UltiMaker printer definitions came from either a member of the community (who owns one), or from the manufacturer itself.

The Ender 5, 5 Plus, and 5 S1 are supported.  If Creality didn't see fit to submit another definition for the 5 Pro it may be because it is close enough to one of the other definitions that it doesn't matter.  You can install an Ender 5 and then go to the Manage Printers and Machine Settings and make any changes that would be required.
The FlashForge printers are different.  I don't think any of the defintions came from FlashForge.  It looks to me like they really want people to use FlashPrint.
There are two plugins in the Marketplace that seem to apply to FlashForge printers.  I don't have one so I don't use them.
[image]

LiesRom, 2024-06-20:
Thank you for your quick reply and explanation. I will have a look at what you suggested.