Ultimaker/Cura #19258: The slicing operation failed with an unexpected error. Please consider reporting a bug on our issue tracker.

ozyapi41, 2024-06-19:
### Cura Version

 5.7

### Operating System

windows 11

### Printer

Anycubic kobra max

### Name abnormal settings

-

### Describe model location

-

### Describe your model

bumblebee mask 

### Add your .zip here ⬇️

[Uploading Bumblebee_Helmet_MK_2.zip…]()

GregValiant, 2024-06-20:
The project / zip file didn't upload correctly and the link just bounces back to this page.

ozyapi41, 2024-06-20:
@
[Uploading Bumblebee_Helmet_MK_2.zip…]()

GregValiant, 2024-06-20:
The posted link address is:
_"https://github.com/Ultimaker/Cura/issues/19258"_ which is this page.

ozyapi41, 2024-06-21:
I don't fully understand why it hasn't been uploaded. Is there anywhere else I can send the file?

GregValiant, 2024-06-21:
It happens every once in a while.  Are you giving it a moment to upload or just exiting the page right away?

One thing you can check - go to the Microsoft Store and download and install "MS 3D Builder".  It has a pretty fair repair utility that runs when you load a file into it.
I always check that the model doesn't have any:  duplicate faces, inverted faces, degenerate faces, and that it is watertight.  A lot of the reports that come in here aren't Cura bugs at all but involve bad models that simply confuse the slicer.

ozyapi41, 2024-06-21:
Yes, I use this application, but there does not appear to be any problem with the program.
[Bumblebee_Helmet_MK_2.zip](https://github.com/user-attachments/files/15924614/Bumblebee_Helmet_MK_2.zip)

GregValiant, 2024-06-21:
That worked.  We really need a project file because sometimes a users settings are part of the problem.

I'm running on Windows 10.  I didn't find any errors in the model and it sliced for me without supports.  Turning on Tree Supports it sliced in 45 minutes and at 50mm/sec is a 169 hour print.
[image]

The problem might be related to Windows 11.  The Cura team will take a look.

ozyapi41, 2024-06-22:
Hello, thank you, I also have a question, how do you make tree supports?

ozyapi41, 2024-06-22:
can you send me this file

GregValiant, 2024-06-22:
I have a rule that I don't provide people with the files I generate.  It is always better for a person to learn how to do it themselves.

Tree supports:
In Cura turn on the "Custom" settings.
To the right of the "Search Settings" box is an icon with three lines on it.  It is the "Setting Visibility" tool.  Set the visibility to "All".
Scroll down to the "Support" section and you will see a setting for "Support Structure".  Set it to "Tree".
There are about 75 settings in the Support section.  Tuning them can take a while.  The "Support Overhang Angle" is the single most important setting and up to about 63° will work with a layer height of .2 and a line width of .4.
"Tree" supports can provide reach from outside a part to an inner surface that needs support.  "Normal" supports slice and print faster.  Always start with "Normal" and go to "Tree" when the particular model indicates it would be a better choice.

ozyapi41, 2024-06-24:
I tried a lot but it didn't work. I need this file but it gives an error in every setting.

GregValiant, 2024-06-24:
Try "Run as Administrator" and see if it will slice in 5.7,0.
Try installing 5.7.1 and see if it will slice.
The last pre-arachne version was 4.13.1.  You could install and try that as well.

Although I volunteer here it is my considered opinion that it is unprofessional to work for free.