Ultimaker/Cura #19259: Coasting and fuzzy skin weirdness.

Stormwing01, 2024-06-19:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Ender 3 pro SE

### Reproduction steps

Turn on coasting (0.15mm volume, 1mm minimum volume, 95 coasting speed) and fuzzy skin (0.02mm thickness, 0.5 point distance) at the same time.

### Actual results

Some coasting lines on the outer wall is suddenly removed/coasting is disabled at random lines. Some lines, at random, do not coast anymore, or coast at much shorter lengths.

### Expected results

coasting should happen at the end of EVERY line, within set parameters. in the project file, coasted lines should all be equal on the model's vertical towers.

### Add your .zip and screenshots here ⬇️

[Cura coasting report.zip](https://github.com/user-attachments/files/15906586/Cura.coasting.report.zip)

GregValiant, 2024-06-20:
Thanks for the report.
Is the printer a Modix 60 or is it an Ender 3?  My printer is an Ender 3 and it isn't capable of a .02mm variation in Fuzzy Skin.  Cura will put that amount into the calculation, but it won't happen in real life at the nozzle.

At 100% flow an extrusion of 1mm³ at a layer height of 0.10 and a Line Width of 0.40 will be 60.132mm long so at a coasting volume of 0.10mm³ the coast distance will be 6.0132mm long.
Your flow numbers for each feature are all slightly different.  This will cause gaps of varying length in each feature.

Fuzzy Skin breaks a single long extrusion into numerous zig-zag extrusions.  When the slice gets to the towers in this project very few (if any) extrusions are 6mm long so the coasting volume criteria isn't being met.  It is causing Coasting to be ignored on many layers after the base completes.

I'll leave the bug label on this and the Cura Team will take a look.

Your settings and the small physical size of the model are having a definite effect.  I have to ask why you are bothering with "Fuzzy Skin" at such a small offset.  With it turned off the model looks much as it would with it turned on at 0.02 Fuzzy Skin Thickness, but with it turned off, Coasting looks correct.

I've added a larger model here.  The same settings affect both models but there are obviously different results.
[image]

This is with Fuzzy Skin turned off.
[image]

This is with Coasting and Fuzzy Skin turned on at their default values for the "Draft" profile.
[image]

Stormwing01, 2024-06-20:
> Thanks for the report. Is the printer a Modix 60 or is it an Ender 3? My printer is an Ender 3 and it isn't capable of a .02mm variation in Fuzzy Skin. Cura will put that amount into the [...]

it is an Ender 3. the stock motherboard (1.1.5 silent edition) is flashed with custom th3d firmware.

0.02mm fuzzy works just fine for me and does show up when printed, so that's rather interesting... but for this bug, you could just use any value, I guess. The reason for the small fuzzy skin was because I was just playing around and I liked the texture that 0.02 gave me. 0.05 was too rough, so I just made it smaller.

I have changed the coasting minimum volume criteria from 0mm all the way up to 20mm, and it has absolutely no effect (besides 20 removing all coasting), but even at 0, some coasting walls still cont coast at all, while others do.

Stormwing01, 2024-07-17:
Bit of a small update: I just noticed that this doesn't just happen with fuzzy skin turned on, but even with it tuned off.

even with fuzzy skin turned off, a coasting move is sometimes entirely skipped or shortened on the inside of rounded curves, such as the inside of some circular holes
[image]