Ultimaker/Cura #19260: Cura Save As Slicing Profile Error

Firedogs2x, 2024-06-20:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Modix

### Reproduction steps

I created a slicing profile to set flow percentage. It has one wall and no top or bottom layers.
I take that profile and save it as a new profile which includes four walls and four top and bottom layers.
Issue only happens when I convert my flow profile to a normal profile.

### Actual results

When I slice using the newly created profile the parts are all messed up. Missing layers, missing infill, and surfaces that are not suppose to be there.

### Expected results

A sliced part which has all of the proper quantity of walls, top and bottom layers, and any specified infill. Plus no extra surfaces were there should be none.

### Add your .zip and screenshots here ⬇️

[Shock Mount Rev A.zip](https://github.com/user-attachments/files/15910162/Shock.Mount.Rev.A.zip)
[Cura Slicing Profiles 06-19-2024.zip](https://github.com/user-attachments/files/15910240/Cura.Slicing.Profiles.06-19-2024.zip)

GregValiant, 2024-06-20:
Thanks for the report.
In the "Shock Mount" project:
You have "Spiralize Outer Contour", "Adaptive Layers" and "Enable Bridge Settings" turned on.
Spiralize and Bridge Settings are mutually exclusive for this little model.
You cannot use Spiraliize for multiple models, or for a single model that develops islands as it prints.  Both conditions are present with this project.  Spiralize requires that the nozzle can move without an extrusion stopping.  That cannot happen if there are islands on a layer and there are always islands with separate models.
"Adaptive Layers" has been set with a maximum variation of 0.10 but your layer height is only 0.10.  That is causing conditions like this with wildly varying layer heights.  (This is with Spiralize turned off).
[image]

Using the stock "Fine" profile the slice looks like this:
[image]

Looking at both the profiles you posted, they both have the same problem.  The settings are inappropriate for this model(s).

 The problems I see are the result of Cura settings which are simply doing what they were designed to do.  The slice looks terrible but it's "Expected Behavior" given the combination of settings.  I'll remove the bug label from this.  Let me know if you feel that is premature.

Firedogs2x, 2024-06-20:
Hello, thank you for the quick reply.
I must have missed those settings when I stepped through the profile.
I apologize and will close this issue.