Ultimaker/Cura #19263: CURA5.7.1和5.7.2都无法切片

08716, 2024-06-21:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

DIY

### Name abnormal settings

不会

### Describe model location

不会

### Describe your model

使用Ocra可以正常切片，CURA5.6也可以。但5.7.1和5.7.2都无法切片

### Add your .zip here ⬇️

你好，我不知道这个文件能不能打开（但我试了十几个模型，包含方块测试魔性之类的都无法切片）。我希望下面的信息能有一定的帮助
如上所述，5.6能正常使用但5.7.1和5.7.2都不行（我已经一个多月没怎么用CURA了但问题还是没有修复）在下载5.7.1和5.7.2后能正常的载入我的打印机配置，第一次进入也有提示这个版本更新了什么。但切片都无法成功）
期待你的回复，谢谢

08716, 2024-06-21:
文件可以导入进CURA，也可以导出3MF格式的文件
但无法切片

GregValiant, 2024-06-21:
Thanks for the report.
Sometimes right-clicking on the desktop icon and selecting "Run as Administrator" can allow this bug to be bypassed.

The Cura team will take a look.  There have been other reports of this behavior on some Windows systems.
I'll remove the "Slicing Error" label as it doesn't sound like it ever gets around to slicing.

08716, 2024-06-22:
你好，感谢你的回复。你说的以管理员身份运行我试过了，还是无法切片

08716, 2024-06-24:
请问你们使用QQ或者微信吗

GregValiant, 2024-06-24:
I don't use any social forums and so I have no idea what those are.
You can use a remote storage site (like Google Drive) but you must give permission for anyone to download the file.  I see no reason why I must open an account somewhere just to troubleshoot a project.  I receive more than enough advertising as it is.

08716, 2024-08-26:
你好，在使用5.8版本的Cura后无法切片的问题解决了。但每次关闭Cura后再打开都需要重新去浏览器登陆我的账号，请问可以在哪设置保持我的账号一直是登入状态吗？

08716, 2024-08-29:
期待你的回复，感谢