Ultimaker/Cura #19264: [5.7.2] Undesired Preheating Sequence added before "Start G-code"

khani3s, 2024-06-21:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Custom Ender3 Pro with Klipper

### Reproduction steps

* Configure custom start and end G-code sequences for the printer and extruder.
* Clean Start-Gcode of Printer and Extruder 1
* Add a custom START_PRINT G-code command with temperature variables to Printer:

```
START_PRINT BED_TEMP={material_bed_temperature_layer_0} EXTRUDER_TEMP={material_print_temperature_layer_0} CHAMBER_TEMP={build_volume_temperature} PRINT_MIN=%MINX%,%MINY% PRINT_MAX=%MAXX%,%MAXY%
```

Slice a model using Cura 5.7.1 and note the generated G-code.
Slice the same model using Cura 5.7.2 and compare the generated G-code.

### Actual results

In Cura 5.7.2, the generated G-code includes additional preheating commands before the custom START_PRINT command:

```
(...)
;Generated with Cura_SteamEngine 5.7.2
M104 S210 
M105 
M109 S210
M82 ;absolute extrusion mode
START_PRINT BED_TEMP=60.0 EXTRUDER_TEMP=210.0 CHAMBER_TEMP=0 PRINT_MIN=8.355,119.078 PRINT_MAX=44.733,155.454
(...)
```

### Expected results

In Cura 5.7.1, the generated G-code includes the custom START_PRINT command without preheating commands:

```
(...)
;Generated with Cura_SteamEngine 5.7.1
M82 ;absolute extrusion mode
START_PRINT BED_TEMP=60.0 EXTRUDER_TEMP=210.0 CHAMBER_TEMP=0 PRINT_MIN=8.355,119.078 PRINT_MAX=44.733,155.454
(...)
```

The generated G-code in Cura 5.7.2 should be similar to the G-code from Cura 5.7.1, where the custom START_PRINT command appears without additional preheating commands.

### Add your .zip and screenshots here ⬇️

.

khani3s, 2024-06-21:
I think that is related to with the
" Improved behaviors for start temperature for multiple extruders"

yazz007, 2024-06-21:
IMHO There is unnecessary wait for hot-end temperature, in the same time bad heating can start, so both starts at the same time.

BearBBQ1882, 2024-06-21:
This new hot end Gcode before the start Gcode in version 5.7.2 is most certainly erroneous. For those of us with UBL and ABL, the hot end is going to heat up before probing and leave little blobs of filament all over the build plate.

GregValiant, 2024-06-21:
Thanks for the report.
This is a duplicate.  Please see bug report #19204 .  I wrote a post processor to address the problem.

Unzip the attached file and put the "CuraPrependBugFix.py" into the Cura installation folder.  On Windows it would be something like:
"C:\Program Files\UltiMaker Cura 5.7.2\share\cura\plugins\PostProcessingPlugin\ **scripts**

It could also go into the Configuration Folder which on Windows would be something like:
"...\AppData\Roaming\cura\5.7\ **scripts**

After a restart of Cura in will be available under "Extensions / Post Processing / Modify Gcode" and then "Add Script".  It simply deletes the extra temperature lines that are being mistakenly added in front of the StartUp Gcode.

[CuraPrependBugFix.zip](https://github.com/user-attachments/files/15933330/CuraPrependBugFix.zip)

The Cura team is aware of this so I'll go ahead and close this report in favor of the first one.

closed 2024-06-21 by GregValiant