Ultimaker/Cura #19265: Won't load models

CarsRFun, 2024-06-21:
### Cura Version

5.7.0

### Operating System

windows 10

### Printer

Creality Ender 3 V2

### Reproduction steps

Whenever I try to upload a file to slice it says 'loading model failed.'

### Actual results

Whenever I try to upload a file to slice it says 'loading model failed.'

### Expected results

It should just download the file like it used to.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-06-21:
Is this for [thing 6652740](https://www.thingiverse.com/thing:6652740) "Self Balancing Bird"?

If so, more information is going to be needed. I'm unable to replicate the problem on my local install, which means there could be something system specific going on. In particular a log file is likely to be beneficial. You can find the log file by clicking **Help**, then **Show Configuration Folder** and looking for the file **cura.log**.

closed 2024-07-05 by github-actions[bot]

zebecasse, 2024-08-02:
[cura.log](https://github.com/user-attachments/files/16466399/cura.log)

I have the same issue, i just update to the new version of cura (from 5.7.X to 5.8), and the direct link from thingiverse don't work, i send you the log of the clic, i want to reach this project : https://www.thingiverse.com/thing:6610030/files

But in fact i have the issue in other projetcs too, so i don't thin the issue is on thingiverse.

I try to uninstall reinstall, but not better, seems that something is wrong on my devide, but can't find what :(

I you can help me please ^^

nallath, 2024-08-02:
> [cura.log](https://github.com/user-attachments/files/16466399/cura.log)
>
> I have the same issue, i just update to the new version of cura (from 5.7.X to 5.8), and the direct link from thingiverse [...]

If you try again it should work now