Ultimaker/Cura #19267: Merge Models does not work well

jorgeagalaz, 2024-06-21:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Creality CR5 

### Reproduction steps

I have just install cura 5.7.1. When I try to merge two models, Cura doesn't work well (The two models are not in the same relative position as in the CAD model). In the cura previous version this doesn't happen.

### Actual results

I have just install cura 5.7.1. When I try to merge two models, Cura doesn't work well (The two models are not in the same relative position as in the CAD model). In the cura previous version this doesn't happen.

### Expected results

I have just install cura 5.7.1. When I try to merge two models, Cura doesn't work well (The two models are not in the same relative position as in the CAD model). In the cura previous version this doesn't happen.

### Add your .zip and screenshots here ⬇️

[image]

jorgeagalaz, 2024-06-21:
> Thanks for the report.
> Without a project file we have nothing to investigate.
> Are you exporting the CAD models separately and as 3mf files?

hello
I attach the files

BR

GregValiant, 2024-06-21:
Github rules are that the files must be uploaded directly to the site.  They do not come in via email.
Zip the files and post the zip folder here.

jorgeagalaz, 2024-06-25:
[20240618_Mikel_Ugarte_Rev01.zip](https://github.com/user-attachments/files/15967458/20240618_Mikel_Ugarte_Rev01.zip)

jorgeagalaz, 2024-06-25:
[20240618_kk_Rev01.zip](https://github.com/user-attachments/files/15967489/20240618_kk_Rev01.zip)