Ultimaker/Cura #19272: when cutting large models there is a bug

jorgecohen, 2024-06-22:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

artillery hornet

### Describe your model

mantaray 

### Add your .zip here ⬇️

i didnt understand

GregValiant, 2024-06-22:
Thanks for the report.
5.5.0 and 5.6.0 had trouble slicing "organically" shaped models which I assume your Manta Ray is.  It was a well known bug and was addressed in later releases.
I suggest you upgrade to 5.7.1.

I'll mark this as a duplicate and go ahead and close it.  I hope you understand.

closed 2024-06-22 by GregValiant

jorgecohen, 2024-06-22:
Oh there is a new version? OK so I will upgrade

בתאריך שבת, 22 ביוני 2024, 11:59, מאת GregValiant ***@***.***