Ultimaker/Cura #19273: Unwanted G-code heating commands despite temperature variables in custom start code

timhendriks93, 2024-06-22:
### Cura Version

5.7.2

### Operating System

Linux Mint

### Printer

Creality Ender 3

### Reproduction steps

1. Go to the machine settings of your printer
2. Define a start G-code containing the following variables
    - {material_bed_temperature_layer_0}
    - {material_print_temperature_layer_0}
3. Load, slice and save any model
4. Check the first lines of the generated .gcode file
5. Note that Cura added commands to heat the nozzle before the custom start code

The actual and expected results are based on the following custom start code:
```
G92 E0 ; Reset Extruder 
M140 S{material_bed_temperature_layer_0} ; start preheating the bed 
M104 S{material_print_temperature_layer_0} T0 ; start preheating hotend 
G28 ; home 
G29 ; ABL 
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} T0 ﻿; wait for hotend temp
```

Things I noticed:
- The issue started directly after upgrading Cura without changing the existing start code
- The temperature values in the older Cura version were inserted as decimal number (e.g. 200.0 instead of 200)
- The release notes for version 5.7.2 show the following 2 points which could be related to this issue:
   - Improved behaviors for start temperature for multiple extruders
   - Improved preheating logic when using a raft with multiple extruders

### Actual results

```
;Generated with Cura_SteamEngine 5.7.2
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G92 E0 ; Reset Extruder 
M140 S45 ; start preheating the bed 
M104 S200 T0 ; start preheating hotend 
G28 ; home
G29 ; ABL
M190 S45 ; wait for bed temp
M109 S200 T0 ﻿; wait for hotend temp
```

The following lines are unexpected after using the temperature variables above in a custom start code:
```
M104 S200
M105
M109 S200
```

### Expected results

```
;Generated with Cura_SteamEngine 5.6.0
M82 ;absolute extrusion mode
G92 E0 ; Reset Extruder 
M140 S45 ; start preheating the bed 
M104 S200.0 ﻿T0 ; start preheating hotend 
G28 ; home 
G29 ; ABL
M190 S45 ; wait for bed temp
M109 S200.0 T0 ﻿; wait for hotend temp
```

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-06-22:
Thanks for the report.
This is a duplicate of #19204 .  If you go to that page and scroll down you will find a post-processor "CuraPrependBugFix"  that will remove the lines.  The Cura team is aware of the bug and it should be fixed in the next release.

I'll mark this as a duplicate and go ahead and close it.  I hope you understand.

closed 2024-06-22 by GregValiant

timhendriks93, 2024-06-22:
@GregValiant sure, thanks for the quick reply and sorry for not seeing the other issue in the first instance 😊

GregValiant, 2024-06-22:
Yours was about the 5th report of the bug so you aren't alone.

If you read through that other post this was a definite "Oops!!!" moment.  An "Unexpected Consequence" of some other fix.