Ultimaker/Cura #19275: Unexpected slicing error.

Jolnear, 2024-06-23:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Creality CR-10

### Name abnormal settings

Fails with default

### Describe model location

I have moved it around, rotated it, etc.

### Describe your model

It is watertight. I can slice it weirdly if i shrink it 20%. 

### Add your .zip here ⬇️

[cura.log](https://github.com/user-attachments/files/15942245/cura.log)
https://we.tl/t-CfkiAWB3Lv

We are back again with more slicing fails. I upgraded to the newest edition. It will slice if I go 20% smaller. I have tried disabling my antivirus (Windows Defender) and moving it around. Its probably just a me issue but I'm hoping the log fixes something for someone else. It won't print on the standard settings either let alone mine.

GregValiant, 2024-06-23:
Thanks for the report.
This is a generic error code from your log file.
"Backend exited abnormally with return code 3221225477!"
MS 3D Builder complained about errors in the model, but the model sliced OK for me as received.

The Cura team will take a look.

Jolnear, 2024-06-23:
As a weird update, it will slice with low profile but not standard/dynamic/super built in profiles (or my custom).

GregValiant, 2024-06-23:
There might be a tiny error in the model and when the combination of Initial Layer Height and Layer Height is just exactly right the slice hits the error and the crash happens.  That's my theory anyway.  

If there is a small gap in a model and it falls below some minimum within Cura, then the gap gets ignored.  If you scale up the model then the gap gets scaled up as well and can become large enough that the slicer notices, gets confused, and quits.  That condition I know happens.

Jolnear, 2024-06-28:
I will say it seems to be more associated with the size. Small models are way more likely to slice. Anything that starts to get above 32 mm seems to struggle. If you guys have a model that you think is good that is larger I could try and slice that if it helps you diagnose any problems and help fix them. I would like to be able to slice without issues haha.

GregValiant, 2024-06-28:
Nobody likes buggy software.  I don't, you don't, and the developers don't.
I'm retired and this is a hobby for me.  I spent a lot of years as a tool designer and as a forensic engineer which is what got me sucked into doing the triage here.  I learn a lot doing this.

I've been accused at various times of being given "special" releases of Cura, and of having access to some sort of "magic" software that fixes everything.  I have noticed that I don't have near as many problems slicing as some people do.  Why that is I have no idea.  This is a second hand laptop that I got for cheap, and my printer is an Ender 3 Pro that I got for cheap.  I don't have multiple monitors, raspberry PI's, Octoprint...nothing fancy at all.  Things just work.

Consider that hundreds of thousands of printer owners use Cura.  There are currently about 3 bug reports per day coming in and they are often (75-80%) because of bad models or inappropriate user settings.  I think that is a pretty good record.  Knowing that you are in the 25% minority is no comfort.  People just want to print.

Your generic "won't slice" problem could be because Cura didn't quite install correctly, or there is some sort of interference with your Windows installation.  The problem came up a lot more with the 5.5 and 5.6 versions of Cura.  It's something I've never been able to duplicate but it certainly happened a lot.  It does appear to be system specific which makes it really hard to troubleshoot from afar.

All the bug reports are taken seriously but getting to them is a matter of priorities.  Resources at UM need to be allocated and a priority list made up.  I don't know how many people are on the Cura Team, but it isn't many.  Along with Cura there are the Github and UM Forum websites, Thingiverse, firmware for several different printer models, 'How To' articles to write and update, and then there is Cura itself which is arguably the most difficult piece of software.
5.7 was a nice change from the problem-ridden 5.5 and 5.6 versions.  We are probably not far from 5.8 and hopefully it will be another step in the right direction.

Jolnear, 2024-06-28:
Sorry yeah not trying to complain just posting things that I notice in the hopes it gets fixed for future and/or if people have anything to try that maybe I haven't. I understand I'm a real fringe case here haha. Tbh idk how I would tell if model is good or not so didn't want to say it was (it say's its watertight but dunno if that means good model) and if someone had a "good model" then wasn't sure if that would help narrow things.

GregValiant, 2024-06-28:
Prusa has some sort of deal with AutoDesk and the NetFabb repair utility is available from within PrusaSlicer.  It's pretty good.
I use MS 3D Builder for repair.  The utility is good and the app itself is very good for altering STL files.  It isn't intuitive, but it does work well.
One website that I still use is [https://formware.co/OnlineStlRepair](https://formware.co/OnlineStlRepair).  It gives a report regarding what it may have found that was wrong with a model.  That's nice to post here.  The site has a file size limit of about 25mb.

Jolnear, 2024-06-29:
Going to try and download Cura on wives PC and see if it will slice on there. Did a complete uninstall and deleted local files etc. I did a complete clean reinstall on admin mode, it still failed. The model was put through the stl repair and that didn't change anything. If it works on my wife PC, I might try to reformat my PC. IDK if it's a driver or association with hardware but we will keep trying things. If you have any ideas for me to try I'm down. I was even desperately thinking about upgrading to win 11 haha. 

[cura.log](https://github.com/user-attachments/files/16042431/cura.log)

GregValiant, 2024-06-30:
It appears that there is some sort of conflict on your laptop.  From the log file:
_"Backend exited abnormally with return code 3221225501!"_
I don't know that particular code.  It could be just a "generic" error code that occurs when Cura can't figure out what broke.

The only Python coding I do is for use as Cura post-processors so I didn't need an extra install of Python.  I don't know if one was bundled with Visual Studio or not as I don't use it for Python scripts.  Notepad++ works good enough for me.  I have not had any conflicts and everything appears to exist together OK.

Jolnear, 2024-09-23:
I FOUND THE FIX. I scrounged through github here and found an old post while ago stating their i9 13500k wasn't slicing and it had something to do with the cores and multithreading. I didn't want to deal with the bios and trying to figure that out but they had mentioned if you go into task manager and set manual affinity and select cores it should work. Low and behold it went from not slicing and crashing to completely slicing. So it must be something specific with this cpu. No idea if that is on cura to fix but maybe the solution to recommend if anyone else comes by with a i9.