Ultimaker/Cura #19281: Cura  Crashing

jzagyi13, 2024-06-25:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Creality  Ender 3 Pro

### Reproduction steps

Select Extentions/Auto Towers. I can select and run any of the Auto Towers on the first go round but when I try to select another auto tower, cura crashes.

### Actual results

Cura crashed when I select Auto Towers after I select it successfully the first time.

### Expected results

I should be able to open a tower under Auto Towers and select a second time without Cura crashing.

### Add your .zip and screenshots here ⬇️

No files to add.

GregValiant, 2024-06-25:
Thanks for the report.
I'm on Windows 10 and my own printer is an Ender 3 Pro   I can't duplicate the problem.  Every time I select a different "tower" it comes in and slices.
You may want to post a bug report on the authors GitHub page.  The plugin was developed by an independent contributor rather than by UltiMaker.
Use the "Help | Show Configuration Folder" command in Cura to locate the Cura.log file.  Zip it and post the zip folder here.  Do not use email.....post the file directly to this site.

jzagyi13, 2024-06-25:
Sorry to waste your time......uninstalling and re-installing the plug-in did the trick.
I should have done that first!
Thanks for your time!!!

closed 2024-06-25 by jzagyi13