Ultimaker/Cura #19284: add two simple gcode commands for the mecreator2

ideechaniz, 2024-06-25:
### Is your feature request related to a problem?

Cura works well on the mecreator2 printer from geetech but the default gcode settings lack a very simple and small feature:

At the beggining the first gcode should be: M42 P6 S255
At the end, the last gcode should be: M42 P6 S0

That basically turns on the leds while printing and turns them off when finishing.

### Describe the solution you'd like

At the beggining the first gcode should be: M42 P6 S255
At the end, the last gcode should be: M42 P6 S0

### Describe alternatives you've considered

I put it on manually but since it is an old printer it would be nice to have it by default because documentation related to old printers is not readily available.

This old printer model has sold many ultimakers (2 and 3 models) back in the days so don't see it as a competing printer ;)

### Affected users and/or printers

mecreator2 from geetech

ideechaniz, 2024-06-25:
The place to put it in cura probably would be here:

/resources/definitions/geeetech_MeCreator2.def.json

GregValiant, 2024-06-25:
This should be handled in your startup gcode.  There is a "light on" line in the definition file.  Changing it will require someone who owns one of those printers to submit a pull request and at that point the change would affect everyone who owns one of those printers.  It's possible that some machines require the M42 line rather than the M2012 line that is there now.

When I add a MECREATOR2 to Cura this line is in the startup Gcode:

**M2012** P8 S1 F100 ; ON Light

There is no "OFF Light" command in the Ending Gcode.  Looking at the syntax I would guess it would be "M2012 P8 S1 F0"
If you require a different command then simply change that line.  You could add the M42 "Off" line to your Ending Gcode.

This is something that must come from Geeetech or an owner of one of those printers.  No one else has any way to tell which way would be correct.

I'll go ahead and close this as it is a change you must make yourself if the M42 command works for you rather than the M2012 line that is already in the definition.

closed 2024-06-25 by GregValiant

ideechaniz, 2024-06-25:
> Hi, thanks for the reply.
>
> I have a mecreator 2 and I can confirm that the line
>
> *M2012* P8 S1 F100 ; ON Light
>
> Doesn't make any effect on the light and it prints with the light off.
>
> the one I put does turn on and off the light.
>
> Anyway thanks for the reply, I just wanted to contribute what I did to use
> the light he way it was supposed to work on this old printer.
>
> Best regards,
>
> Iker.

On the official manual on page 26:

https://www.geeetech.com/Documents/Me%20Creator%202User%20Manual.pdf

5.1.4 LED Light Test
There is an LED strip in the MeCreator2, you can control the light by
sending corresponding command.
Turn on： M42 P6 S255 (any number between 1-255, the greater the number is,
the lighter it will be)
Turn off： M42 P6 S0

Best regards,

Iker

ideechaniz, 2024-06-25:
> It appears that the definition files for Geeetech printers come from
> Geeetech. You will need to go back to them and have them submit a
> definition change as a Pull Request.
> In the meantime you can go to Manage Printers / Machine Settings and
> change your StartUp Gcode and Ending Gcodes to include those lines.

Ok I will do that.

i am impressed by the way you dealt with this in a good way, thanks.

Best regards,

Iker.