Ultimaker/Cura #19285: Lightning Infill produces incorrect results when "Infill Support" is checked

Nogibator99, 2024-06-25:
### Cura Version

5.7.2

### Operating System

Mac OS 13.3.1

### Printer

Anycubic Kobra 2

### Reproduction steps

1. Set "Infill Pattern" to some value other than "Lightning"
2. Check the "Infill Support" box
3. Set "Infill Pattern" to "Lightning"
4. Observe that the "Infill Support" box is not visible anymore, which probably makes sense, since "Lightning" infill mode takes care of supports
5. Slice

### Actual results

Some parts of the infill are hanging in the air

### Expected results

The result should not depend on the invisible checkbox

### Add your .zip and screenshots here ⬇️

https://github.com/Ultimaker/Cura/assets/13188098/349f65b6-6302-4f18-812b-fc9f0112c78c
https://github.com/Ultimaker/Cura/assets/13188098/3a3e1142-4d6e-4e37-a6f6-57110bdbd925

GregValiant, 2024-06-25:
Thanks for the report.
We ask for Project Files because they are necessary for an investigation.  I made up a spherical model but I cannot duplicate the problem.
Please post a project file.

Nogibator99, 2024-06-25:
Thanks for getting back!
Sure, here it is.
[Bro.3mf.zip](https://github.com/user-attachments/files/15979293/Bro.3mf.zip)

The problem gets more apparent when you look at the infill closer to the top of the model. You can set infill density to 10 to make the effect more clear.

Attaching a video that can hopefully convey what I'm doing.
https://github.com/Ultimaker/Cura/assets/13188098/62faffa2-b0c3-4e11-b342-19d0675029d4

GregValiant, 2024-06-25:
This is your project with Infill Density at 10% and "Infill Support" was turned on before switching to "Lightning" infill.  Those vertical gaps should not be there and are a 100% failure problem.
[image]

That might be why Infill Support was disabled for Lightning but even though the setting is hidden, it still seems to be having an effect.  This reminds me of what was happening with some other settings in the Support section.  The settings were being hidden due to the values of other settings, but they were still having an effect.

@HellAholic and the Cura team will take a look.

Nogibator99, 2024-06-25:
Yes, this is precisely what's going on. That's great, thanks a lot for your effort, really appreciate it!