Ultimaker/Cura #19290: Skin bottom pattern in bridging settings.

Rwide, 2024-06-26:
### Is your feature request related to a problem?

When printing bridges it can sometimes happen that anchoring points fail to attach.

### Describe the solution you'd like

To eliminate/reduce this I would love to have the option to choose continous line (like zigzag) pattern for the bottom layer of the bridge, like in the picture. I had to dissable the "bridge settings" and set the "extra skin wall count" to 0. 

A setting like "skin overlap percentage" could then be used to control the strength of the anchoring points. I think this would be a great addition to the bridge settings for people with bowden extruders, people with uncalibrated preasure advance settings and others.
[image]
 

### Describe alternatives you've considered

I did try "connect infill lines" but it doesn't seem to affect the skin ( at least not in the preview).

### Affected users and/or printers

 people with bowden extruders, people with uncalibrated preasure/linnear advance settings and others. 

### Additional information & file uploads

I used these settings to achieve the skin pattern:

Bridge settings=deactivated
Extra "infill wall count=0
Skin overlap percentage=a reasonable amount
Bottom pattern=zigzag

The problem is that this affects all skins. Not just the bottom of bridges. One would also loose control over other bridge settings because the bridge settings have to be dissabled to achieve this pattern.

smartavionics, 2024-06-26:
My Cura 4 variant already provides this capability (along with lots of other tweaks/features). If you want to try it, go to https://github.com/smartavionics/Cura/releases.

[image]

Rwide, 2024-06-26:
@smartavionics That it awesome! Thank you. Unfortunately Cura 4 has the old (not)random z-seam functionality wich is why i abbandoned Cura a couple of years ago. I believe it was fixed in Cura 5.2 release..

smartavionics, 2024-06-26:
Well, I fixed the non-random random z-seam issue in my own way as you can see here...
If you do try my release and have any issues, please open a new issue at my repo rather than here, thanks.

[image]

Rwide, 2024-06-26:
Very nice work! I will definitely try it.

Rwide, 2024-06-27:
Just for clarification. This is how bridges should be made imo. 
[image]
Picture is from smartavionics Cura fork..

smartavionics, 2024-06-28:
FWIW, I still use lines pattern rather than zigzag for bridge skins myself.