Ultimaker/Cura #19295: Conan fails on building CuraEngine

pskowronskiTDx, 2024-06-27:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

None

### Reproduction steps

Hello, I have been following the instructions on "Building Cura" but unfortunately I am unable to install dependecies with conan. The process fails at building the CuraEngine. I couldn't find any solution in already reported issues. Also, I have troubleshooted the issue as described on instructions page but it didn't help. 

### Actual results

ConanException: Error 1 while executing cmake --build "C:\.conan\ddae59\1\build\Release" -- -j16

### Expected results

Successfull installation of the dependencies.

### Add your .zip and screenshots here ⬇️

[image]

...

[image]

wawanbreton, 2024-06-28:
Hi @pskowronskiTDx,
Sorry for this, we had some bugs recently with our build system. I actually made a fix this morning that should help. Can you please clean your conan environment and try again ? If it still doesn't work, please join the whole conan output.

pskowronskiTDx, 2024-07-01:
Unfortunately, it keeps failing the exact same way. Please find here a full conan log:
[install.log](https://github.com/user-attachments/files/16050352/install.log)

wawanbreton, 2024-07-01:
Oh, that is annoying. We also have similar issues on our side, so I'll make a fix and keep you posted.

wawanbreton, 2024-07-01:
@pskowronskiTDx It should be working for you now. I advise you to:
1. Clean your Conan cache
2. Update your branches against their respective `main`
3. Remove any `build`/`venv` directory in your work directory
4. Restart the dependencies install

pskowronskiTDx, 2024-07-01:
Thank you, it looks like the installation process has completed successfully now. Currently, I am facing this issue: https://github.com/Ultimaker/Cura/issues/19299, but it seems to be solveable by VS version downgrade.

wawanbreton, 2024-07-01:
I'm happy to hear that you got further :)
For the other issue, if you can indeed downgrade VS easily, that is the quickest solution. Otherwise you can rebuild arcus/pyArcus but that can lead you into new troubles...

closed 2024-07-02 by wawanbreton

wawanbreton, 2024-07-02:
Hmm this is weird, did you do a `conan profile new default --detect --force` before reinstalling dependencies ? Maybe it still used the previous VS version.

pskowronskiTDx, 2024-07-02:
Yes, I have repeated all of the steps. Currently I don't have any other installations of VS 2022 other than 17.8.4.

wawanbreton, 2024-07-02:
Ok, I don't really know why it is failing then. I would advise to rebuild arcus+pyArcus yourself and use it as an `editable` conan package, but that can be difficult. Maybe @rburema has a better opinion ?

vjvarada, 2024-07-04:
> Okay, so I have downgraded my VS 2022 to version 17.8.4, cleaned up old output and reinstalled conan dependencies. Apparently, in my case these actions did not fix pyArcus issue. I keep running into the same error.

TBH, not sure if it was only VS 2022, or other components. Because I had successfully compiled in April, I reverted my whole environment to what I was using in Apri because I noticed that 5.7.0 which has successfully compiled before was also throwing the same tantrum :
1. Visual Studio 17.8.4  get professional release [https://learn.microsoft.com/en-us/visualstudio/releases/2022/release-history]
2. Cmake 3.27.8
3. Ninja 1.11.1
4. Conan 1.60.2
5. sip 6.8.3
6. python 3.10.11

I also cloned Ultimakers conan-config repo and reverted back to the last commit in April just for good measure. any one of the above could have fixed the issue. Any one or multiple of the above could have fixed it.

rburema, 2024-07-04:
I'll try to look at this later today as well (and I'll notice that I've only partly had to recompile things for a while, so my current config might not be 100% save) but the versions I have on my local machine (as opposed to the github runners) of all of those things currently (while CuraEngine compiles for me) is:

- Visual Studio Community 2022 (64-bit) -- ver. 17.9.5
- CMake 3.26.3
- Ninja 1.11.1 (same as OP)
- Conan 1.62.0
- sip -- I'll look into this if needed, but the version I'm using isn't the one reported. Probably 6.5.0, since that's the minimum needed according to pyArcus and I didn't upgrade for a long time
- python 3.10.4

When looking into which sip version I use for the working build, I also found the following issue, that seems to be at least related (haven't looked to deep into that yet) -- the suggestion there is that the python-path might be incorrectly configured: https://github.com/Ultimaker/Cura/issues/17717#issuecomment-1890598828

vjvarada, 2024-07-04:
Im able to build curaEngine seperately, but get the following error when compiling cura:
```
ERROR: cpython/3.10.4@ultimaker/stable: Error in build() method, line 425
        self._msvc_build()
while calling '_msvc_build', line 394
        msbuild = MSBuild(self)
        ConanException: Using the wrong 'MSBuild' helper. To use MSBuildDeps, MSBuildToolchain you should use 'from conan.tools.microsoft import MSBuild'. Set environment variable CONAN_DISABLE_STRICT_MODE=1 to override this check (should only be used to build old packages).
 ```
[image]

I could compile it yesterday, but for some reason isn't working today. setting CONAN_DISABLE_STRICT_MODE in environment variables just gives `ConanException: Error 1.........` while executing eventually. Tried setting to the versions you are using @rburema 

https://github.com/Ultimaker/Cura/issues/17717 Is something I ran into earlier, but the solution on the issue fixed it to run cura post building. But now I'm back to being stuck on the compilation step.

rburema, 2024-07-09:
@vjvarada I saw a notification of a reply in my mail, but github seems to have eaten it -- or you may have deleted the response. Does this mean this isn't an issue anymore or is something else going on?

pskowronskiTDx, 2024-07-10:
Here is my current configuration. I have updated VS as downgrading it hasn't help.
 
- Visual Studio Professional 2022 - 17.10.3
- CMake 3.26.3
- Ninja 1.11.1
- Conan 1.60.2
- sip 6.8.3
- python 3.10.7

Conan installation fails as follows:
[image]

Full conan install log:
[conan_log.txt](https://github.com/user-attachments/files/16157507/conan_log.txt)

runruiliu, 2024-08-20:
> Ok, I don't really know why it is failing then. I would advise to rebuild arcus+pyArcus yourself and use it as an `editable` conan package, but that can be difficult. Maybe @rburema has a better opinion ?

Maybe you can explain in detail how to rebuild arcus and pyarcus. I also encountered the same dll loading problem and don't know how to solve it. I'm looking forward to your reply.

runruiliu, 2024-08-20:
> @vjvarada I saw a notification of a reply in my mail, but github seems to have eaten it -- or you may have deleted the response. Does this mean this isn't an issue anymore or is something else going on?

Is there a solution to this problem? I've been stuck here for a long time, looking forward to your solution