Ultimaker/Cura #19298: Issue by bridges on test cubes

darkfrei, 2024-06-27:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Anicubic Kobra 2

### Name abnormal settings

Bridges

### Describe model location

default

### Describe your model

Two cubes

### Add your .zip here ⬇️

Enable Bridge Setting: true

The top layer is impossible
[image]
[testCubes.zip](https://github.com/user-attachments/files/16018539/testCubes.zip)

smartavionics, 2024-06-27:
Yes, because your cubes are hollow and the width of the walls is less than three line widths.

Incidentally, my Cura 4 variant has an option to remove the floating inner walls and so it produces this...

[image]

You can find my releases at https://github.com/smartavionics/Cura/releases.

darkfrei, 2024-06-27:
I cannot find this setting in the Cura 5.7.1, sorry:

[image]

smartavionics, 2024-06-27:
That setting doesn't exist in UM Cura.

Unless you really need your model to be hollow, it's generally better to create solid models.

GregValiant, 2024-06-27:
I'll throw my 2 cents in.
@darkfrei it appears that you are trying to fool Cura into thinking that those blocks are hollow, but the models aren't hollow.  They don't require bridging.  If you make them hollow then the results can be adjusted for in the settings.  The way they are right now - if you increase the "Skin Overlap" to 50% and the "Wall Line Count" to 1, I don't think it would fail.

I hollowed this block out and it has 1mm walls.
This is with the Wall Line Count at 1 and this works.
[image]

But with the wall line count at 6 the inner portion isn't attached to anything so it will fail.
[image]

This is a different model.  It should have support but I've turned on bridging instead.  This is with the Wall Line Count at 6.
[image]

Because there is a lot of interior model on the right, the buttress completely holds the bridge.
But on the left side - that "Wall Line Count" of 6 is almost as wide as the feature the bridge is resting on and so it's barely hanging on.  It might not fail, but it's certainly "iffy".
Increasing the Wall Line Count to 10 you can see that now the bridge will absolutely fail because the setting change is forcing the inside area to air-print.
[image]

My point is that the settings you use, and the geometry of the model are always going to play into how the toolpaths turn out.  Not every niche situation can be considered a bug.
@smartavionics has put a fix in place in his fork of Cura.  I use his fork for some of my own printing.  It use it exclusively for Spiralize.
What I'm seeing here is "Expected Behavior".  It's just not the behavior _you_ expected.

I'll leave the bug label on this and the Cura team will decide if they want to incorporate a change.

smartavionics, 2024-06-28:
The cubes are hollow...

[image]

darkfrei, 2024-06-28:
Cubes are hollow: 0.4 mm walls and 0.2 mm on bottom/top. 
[image]

It's a test model to adjust bridges, I have same issue on the other model, that needs 8 hours. And I cannot just set the wall thickness to 1 layer, the big model needs it.

So I've made a test model to make fast prints with bridging over the cube top, I've found, that the enabling the setting "Enable Bridge Settings" changes the covering and the "Bridge Skin Speed" = 10-15 mm/s (instead of 20 mm/s) makes good bridges.
I've restored all other changed settings to be sure, that this setting is the key one and the bridge was not created: the top layer has extra walls, but just the bridge was expected.

My best print, changed settings::
```
Wall Line Count = 1
Extra Skin Wall Count = 0
Enable Bridge Setting = Enabled
Bridge Skin Support Threshold = 100%
Bridge Skin Speed = 10 mm/s
```
[image]

But the issue is by slicing: the Cura must make walls on the other walls or supporting elements, but bridges for elements without supporting, like here:

Enabled "Enable Bridge Setting":
[image]

Disabled "Enable Bridge Setting":
[image]

But Cura slices the top layer by the thin walls wrong , that is the issue.
It must look to the previous layer instead of just setting "Wall Line Count" and "Extra Skin Wall Count".