Ultimaker/Cura #19299: Issue with building from source: ImportError: DLL load failed while importing pyArcus: A dynamic link library (DLL) initialization routine failed

vjvarada, 2024-06-27:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Generic

### Reproduction steps

```
pip install conan==1.60.2
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
git clone https://github.com/Ultimaker/Cura.git 
cd Cura
git checkout 5.7
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
.\venv\Scripts\activate.ps1
python cura_app.py
```

### Actual results

```
Windows fatal exception: access violation

Current thread 0x000009f8 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "C:\.conan\de57bd\1\site-packages\UM\Backend\SignalSocket.py", line 4 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "C:\.conan\de57bd\1\site-packages\UM\Backend\Backend.py", line 11 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "C:\.conan\de57bd\1\site-packages\UM\Qt\QtApplication.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 41 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura_app.py", line 197 in <module>
Windows fatal exception: access violation

Current thread 0x000009f8 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
[... 188 lines cut ...]
    import pyArcus as Arcus
ImportError: DLL load failed while importing pyArcus: A dynamic link library (DLL) initialization routine failed.

Original exception was:
Traceback (most recent call last):
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura_app.py", line 197, in <module>
    from cura.CuraApplication import CuraApplication
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 41, in <module>
    from UM.Qt.QtApplication import QtApplication  # The class we're inheriting from.
  File "C:\.conan\de57bd\1\site-packages\UM\Qt\QtApplication.py", line 21, in <module>
    from UM.Backend.Backend import Backend #For typing.
  File "C:\.conan\de57bd\1\site-packages\UM\Backend\Backend.py", line 11, in <module>
    from UM.Backend.SignalSocket import SignalSocket
  File "C:\.conan\de57bd\1\site-packages\UM\Backend\SignalSocket.py", line 4, in <module>
    import pyArcus as Arcus
ImportError: DLL load failed while importing pyArcus: A dynamic link library (DLL) initialization routine failed.
```

### Expected results

App starts up

### Add your .zip and screenshots here ⬇️

[image]

vjvarada, 2024-06-28:
tries replacing pyArcus and Arcus.dll as mentioned here: https://github.com/Ultimaker/Cura/issues/19192

Cura starts up now, but crashes when loading multiple files
```
Thread 0x0000863c (most recent call first):
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 320 in wait
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 460 in acquire
  File "C:\.conan\de57bd\1\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\.conan\de57bd\1\site-packages\UM\JobQueue.py", line 124 in run
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 1009 in _bootstrap_inner
  File "C:\.conan\6d1c0b\1\bin\lib\threading.py", line 966 in _bootstrap

Current thread 0x00004d2c (most recent call first):
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\Arranging\Nest2DArrange.py", line 123 in findNodePlacement
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\Arranging\Nest2DArrange.py", line 134 in createGroupOperationForArrange
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\Arranging\Arranger.py", line 25 in arrange
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 2188 in _readMeshFinished
  File "C:\.conan\de57bd\1\site-packages\UM\Signal.py", line 357 in __performEmit
  File "C:\.conan\de57bd\1\site-packages\UM\Signal.py", line 328 in __performEmitIndirect
  File "C:\.conan\de57bd\1\site-packages\UM\Event.py", line 218 in call
  File "C:\.conan\de57bd\1\site-packages\UM\Qt\QtApplication.py", line 506 in event
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 1223 in event
  File "C:\.conan\de57bd\1\site-packages\UM\Qt\QtApplication.py", line 420 in exec
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura\CuraApplication.py", line 953 in run
  File "C:\Users\Vijay\Documents\GitHub\Cura\cura_app.py", line 243 in <module>
  ```

vjvarada, 2024-06-29:
@MariMakes do think you could help on this? I'm trying to create a folk with this functionality https://github.com/Ultimaker/Cura/issues/19161

vjvarada, 2024-06-30:
I could fix this by downgrading my Visual Studio installation to 17.8.4 (Which is what I noticed cura actions was using, and not 17.10.3 which is latest) from https://learn.microsoft.com/en-us/visualstudio/releases/2022/release-history

closed 2024-06-30 by vjvarada

runruiliu, 2024-08-15:
Hello, regarding your attempt to replace pyArcus and Arcus.dll, can you elaborate on how to replace them? Although I saw the URL https://github.com/Ultimaker/Cura/issues/19192 you mentioned, I still don’t know how to implement it. I look forward to your reply. Thank you very much!