Ultimaker/Cura #19302: Katana Model not slicing with unexpected error

iamgabrieltv, 2024-06-28:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Creality Ender-3 Pro

### Name abnormal settings

I enabled coasting and changed retraction speed and distance as well as wall thickness and infill. it does slice on the default profile.

### Describe model location

It's centered and slices when i lay it sideways, so 90 degrees on x or y axis.

### Describe your model

It's watertight (according to meshtools) and a fix doesnt help.

### Add your .zip here ⬇️

[CE3PRO_KATANA_SQ_PNP.zip](https://github.com/user-attachments/files/16032646/CE3PRO_KATANA_SQ_PNP.zip)

iamgabrieltv, 2024-06-28:
I forgot to mention im trying to print this model: https://www.printables.com/model/543260-collapsing-katana-v2-print-in-place

GregValiant, 2024-06-28:
Thanks for the report.
I checked the model and it's error-free.  I was able to slice your project with no problem.  If you could duplicate the problem and then immediately zip your Cura.log file (use "Help | Show Configuration Folder" to find it) and post that here it would be helpful.  The log might show what is happening with your system.
The Cura Team will take a look but they will need that log.

iamgabrieltv, 2024-06-28:
So uhm... funny story. I ran the slice again and it crashed like normal and i looked at the log but it was so cluttered and had data from older dates so i deleted cura.log so it could be regenerated so i ran the slice again and it worked. So i have no log of the error. I do remember that it said something about `file %s could not be written to` or smth. Here is the log of the slice that worked if you want it.
[cura.log](https://github.com/user-attachments/files/16036655/cura.log)

GregValiant, 2024-06-29:
If it happens again you'll know what to do.  Cluttered is OK so long as the crash info is in there.  My suggestion was just to make sure that the crash would be one of the last things in the file.  

I'm going to put the "needs info" tag back on this.  One of the things it does is force an automatic closure of a bug report in 2 weeks if there is no response so if you have another problem post the log file.  If not, just leave this be and it will die a natural death.

iamgabrieltv, 2024-06-29:
So as i went to change some print options and re-slice it actually happened again. Here's the log this time.
[cura.zip](https://github.com/user-attachments/files/16039565/cura.zip)

GregValiant, 2024-06-29:
Thanks.  The Cura team will take a look.