Ultimaker/Cura #19303: AppImage fails to start on systems that have mesa built with wayland 1.23.0

prurigro, 2024-06-28:
### Cura Version

5.7.2

### Operating System

Linux

### Printer

Any

### Reproduction steps

Update mesa to a version compiled against 1.23.0 and run the AppImage

### Actual results

When starting the AppImage a GUI popup appears saying "Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers." and the CLI output is "qt.qpa.wayland: EGL not available"

### Expected results

For Cura to start.

---

The full issue (and explanation of what and why) is documented here: https://gitlab.freedesktop.org/mesa/mesa/-/issues/11316

To quote the mesa dev:

> Any AppImage which includes an old libwayland-client.so.0 inside the image, using a newer Mesa from the host, will be affected by this - even the most simple application would fail to start.

The workaround is to extract the AppImage and delete libwayland-client.so.0 so it uses the system version, but the solution would be to ship a new AppImage with an up-to-date libwayland-client.so.0.

Thanks!

### Add your .zip and screenshots here ⬇️

No screenshots are necessary

molove, 2024-07-02:
I have this issue too

5arer, 2024-07-03:
This was bugging me on two PCs, didn't understand how there was not more noise :)
It's unfortunate that using Appimage makes for this kind of issue which is the main idea to prevent
Arch AUR cura-bin image is updated with a fix so one just has to reinstall it from there
Otherwise flatpak version works without problems