Ultimaker/Cura #19304: Slicing failed with unexpected error!

invictus-anic3tus, 2024-06-28:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Anycubic Kobra 2 Neo

### Name abnormal settings

Tree supports caused the crash. It is the basic tree support settings.

### Describe model location

Anywhere, any rotation.

### Describe your model

I have successfully sliced my model before. It is watertight.

### Add your .zip here ⬇️

[A4_Left Stand.zip](https://github.com/user-attachments/files/16033436/A4_Left.Stand.zip)
When using normal supports, as opposed to tree supports, it slices perfectly. It is a tree support error.

GregValiant, 2024-06-28:
Thanks for the report.
Cura 5.5 and 5.6 had slicing issues.  I tried with 5.6.0 and it sliced with no problem.
So I am unable to duplicate the problem in either 5.6.0 or 5.7.1.  A lot of improvements were made in the 5.7 releases and you should update to 5.7.1.

This was a well known problem in 5.6.0 and was worked on and appears to have been fixed in the three versions since then.
I'll mark this as a duplicate and go ahead and close this.  I hope you understand.

closed 2024-06-28 by GregValiant

invictus-anic3tus, 2024-06-28:
OK! Thank you very much!