Ultimaker/Cura #19309: will not let me slice

userError19, 2024-06-30:
### Cura Version

5.4

### Operating System

windows 11

### Printer

ender 3

### Reproduction steps

1. open the file in Cura
2. press slice

### Actual results

It said there was an error

### Expected results

what I said above

### Add your .zip and screenshots here ⬇️

[image]
[image]

Asterchades, 2024-06-30:
There have been numerous fixes made which address this specific error since 5.4 released over a year ago. If nothing else you should at least update to 5.7.2 and try again.

Any fixes would require a new version anyway.

GregValiant, 2024-06-30:
There were well documented problems (numerous bug reports) regarding slicing errors in 5.4-5.6.  Organic shapes were a particular issue.

Follow Asterchades advice and upgrade but I would suggest 5.7.1.
5.7.2 has a bug that can affect the StartUp gcode and requires a post-processing fix (which can be found in bug report #19204 ).  The bug doesn't affect all printers, but affects every printer that has "{material_print_temperature}" and/or "{material_print_temperature_layer_0} in the StartUp Gcode.  That post-processor will fix the bug.

I'll mark this as a duplicate and go ahead and close this as the problem seems to have been fixed in the 5.7.x versions.  I hope you understand.

closed 2024-06-30 by GregValiant