Ultimaker/Cura #19312: MINGDA version of Cura crashes on start (contacted them too)

LilBub, 2024-07-01:
### Cura Version

MINGDA Cura which appears to use 5.3.0 engine

### Operating System

Windows 11

### Printer

Mingda Magician Max2 

### Reproduction steps

1) Installed Mingda's version of Cura.
2) Sliced a Benchie
3) Printed (saved to SD card)
3a) Saved also a Project file, hoping I could open this project in 5.7.2 and import over the printer/printer settings. (That failed, only model would show)
4) Closed Mingda Cura
5) Re-opened Mingda Cura, get a quick black box, then nothing

### Actual results

Crash.  

### Expected results

Should load Mingda's version of Cura.

### Add your .zip and screenshots here ⬇️

FYI - Link to Mingda's version is https://3dmingdaofficial.com/products/magician-max2 (scroll down to "Slicer" link.

Dug around to  AppData\Roaming\MINGDAcura for the error log

[stderr.zip](https://github.com/user-attachments/files/16047645/stderr.zip)

I notified their Support about the issue and crash.  Being a low budget China company, I don't anticipate anything useful from them.  Honestly, I would just like to extract the Printer Profiles from their slicer and import into new Cura.

GregValiant, 2024-07-02:
I'll go ahead and close this as only UltiMaker Cura is supported here.
Other companies (Creality, Mingda, Sovol, etc.) that have their own forks of Cura maintain their own versions.  There is no way to tell how they might have re-written the code for their own use.

closed 2024-07-02 by GregValiant

LilBub, 2024-07-03:
Hi Greg

I created a more useful ticket here ( https://github.com/Ultimaker/Cura/issues/19322 ) 

Working with Slashee the Cow I zipped the folders that contain the definitions, printer settings and slicer defaults for the Mingda 3D printers.  I know the vendor isn't very supportive and I know it's always a frustrating situation for the end user and devs.  Hopefully this helps both sides - you have the data, they end user has the printers in the next build.