Ultimaker/Cura #19314: Printer dropdown remains empty after UI reports successfully adding a Method X

slpenza, 2024-07-02:
### Cura Version

5.6.0-beta.2

### Operating System

MacOS Sonoma 14.5

### Printer

MakerBot Method X

### Reproduction steps

1. Opened Cura app for the first time.
2. Added "UltiMaker printer" by clicking "Sign in to Digital Factory"
3. Acknowledged the addition of two printers: a Method and a Method X. 
4. The printer dropdown does not contain either printer. It is not possible to slice.
5. Clicked "Manage Printers" in printer dropdown. The printer list in Preferences does not contain either printer.
6. Clicked "Add Printer" or "New Printer" from the printer list (I don't remember the button label)
7. Clicked "Sign in to Digital Factory". New web browser tab opens. Cura hangs on "Add printer" modal.
8. Quit & restarted the Cura app.
9. Click "Get Started", "Agree", "Next", then "Ultimaker printer".
10. Clicked "Sign in to Digital Factory" and the app hangs again on "Waiting for new printers"

### Actual results

After clicking "Sign in to Digital Factory" on the "Add printer" modal, a web browser tab opens. Cura hangs on "Add printer" modal after the button label changes to "Waiting for new printers"
[image]

### Expected results

The Method X appears in the printer dropdown so it can be selected.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-07-02:
Thanks for the report.
I would suggest you move from the 5.6 beta pre-release version, to the current 5.7.2 stable version of Cura.
I didn't have any trouble installing the printer as a non-networked printer, but that doesn't equate to a networked printer.  Integration of the Method printers appears to be ongoing.
The Cura team will take a look.

krussell-pem, 2024-07-09:
We have something similar here whereby the S5 and S7 3D printers continuously drop from the list of printers. Both are on the network. We have to readd the IP addresses of the printers for them to come live again. One user is on 5.7 and the other is using 5.7.2. Please look (attached is greyed out printer)
[image]

GregValiant, 2024-07-09:
@HellAholic have you seen this one?