Ultimaker/Cura #19317: Z-hops result in occasional bad layer heights

GregValiant, 2024-07-02:
### Cura Version

5.7.2

### Operating System

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

1. Enable Z-hops at 0.4 height
2. Adaptive Layers off.
3. Layer height 0.2 line width 0.4.

### Actual results

There is (what appears to be) an abhorrent 'Z' move at the beginning of layer 204.
G1 F600 **Z41.1**
;MESH:NONMESH
G0 F7200 X68.233 Y102.366 **Z41.1**
;TIME_ELAPSED:47492.352067
;LAYER:203
G0 X68.233 Y102.366 Z41.3
;TYPE:SUPPORT
G1 F600 **Z40.8**
G1 F900 E3532.9248

G1 F600 **Z41.3**
;MESH:NONMESH
G0 F7200 X82.49 Y103.815 **Z41.3**
;TIME_ELAPSED:47725.485462
;LAYER:204
G0 X82.49 Y103.815 **Z41.5**
;TYPE:SUPPORT
G0 F720 X82.49 Y103.815 **Z41.45**>>>>>>> ????
G1 F600 **Z40.95**  >>>>>>>>>>>>>>>>????
G1 F900 E3614.03371
G1 F671.3 X82.603 Y103.783 E3614.03789
G1 X82.717 Y103.815 E3614.04212
Sometimes this occurs and the layer height gets straightened out mid layer.
G1 X82.49 Y103.815 E3614.08387
G1 F900 E3612.08387
G1 F600 **Z41.5**
G0 F7200 X72.679 Y103.557 **Z41.5**
G1 F600 **Z41**
Other times the condition lasts for the entire layer.  You can see it in the gcode preview below.

All the extrusions are calculated at a 0.20 layer height.
The extrusions that occur during the period where the layer height is 0.15 are over-extruded.  I have had entire layers at the wrong layer height and then the next layer is at 0.25 layer height which makes up for the Z problem, but creates under-extrusion throughout that layer.  The combination of an over-extruded layer and then an under-extruded layer leaves a ridge in the print.
Turning off "Z-hops" makes the problem go away.

The condition does not appear in the Cura Preview.  It does appear when the Gcode file is opened in Cura.
[image]

### Expected results

Each layer should be the same height.

### Add your .zip and screenshots here ⬇️

The project is with no post-processors and no plugins loaded.
[Z-Hop Bug.zip](https://github.com/user-attachments/files/16071195/Z-Hop.Bug.zip)

If I was a betting man I'd bet that this one won't be easy.
This is a standard model I use for a lot of testing.  It's been sliced numerous times.  I stretched it in the Z to see if there would be more occurrences and there were.

This is the preview in Cura:
[image]

This is the preview of the gcode file.  It appears to be missing layers, or model errors, but it isn't and it doesn't.
[image]

Asterchades, 2024-07-03:
This is an issue with the GCode preview, rather than the slice itself. Specifically this is the result of your **Support Z Distance**, which has no longer rounded as of 5.6, being set to something other than a multiple of your **Layer Height**. This is also why you're getting "odd" (figuratively and literally) Z values, as they're being used for the supports rather than the model itself.

Cura really doesn't handle previewing imported GCode files well when there's more than one Z value being printed on a single layer, with the preview struggling to determine what should be bottom of a given layer (though it appears to correctly detected). As such if one of these part-height support pieces exists on a layer the actual model itself will be assumed to start on top of those, rather than with the same "floor" height. This results in the preview squishing that piece of the model and makes it look like a gap is formed in the print. _I've seen Trees do this to up to four consecutive layers at their tips._

The important part is that it's strictly a display issue, and will only occur when reading back GCode. This can (and will) print exactly as expected, matching the slice preview, without any actual layer heights being altered.

Here's a screen cap of the GCode preview with a 0.2mm gap, rather than the 0.25mm gap you're using:
[image]

It's worth mentioning that PrusaSlicer will show the same issue in its preview. IdeaMaker is so far the only preview I've seen which can understand these sorts of layers properly, but I've not conducted exhaustive testing.

GregValiant, 2024-07-03:
Thanks @Asterchades.  I open too many 3mf files for my own good.

Going back over my work I see that the Z is only changing for the interfaces.  The flow of the interfaces does reflect the altered layer height.
I'll go ahead and remove the bug label and close this as it is a preview issue and there was some confusion on the part of an old man.
I hope I understand.

closed 2024-07-03 by GregValiant