Ultimaker/Cura #19322: Please add these Mingda 3D printers into modern Cura - all data provided

LilBub, 2024-07-03:
### Is your feature request related to a problem?

Vendor creates their custom Cura but they do not support / maintain it.  
Thanks to Slashee The Cow he guided me into how to pull those printers from the custom Cura into the current version
https://community.ultimaker.com/topic/45797-pulling-printer-settings-from-vendor-versions-cura/#comment-335516

### Describe the solution you'd like

For the benefit of other Mingda owners, here is the list of 3D printers their Cura version shows.  They have not made these available to the Cura team to add into modern Cura.   I have attached the required resource files so that they can be added.  Here is a screen shot of what that list of machines is.   

Hopefully this will streamline things for the team.  And maybe a bit less grief from consumers angry Cura isn't supporting them (When it is really the manufacturer of the 3D printer!)

[image]
[Cura Files.zip](https://github.com/user-attachments/files/16086749/Cura.Files.zip)

### Describe alternatives you've considered

I tried exporting a 3MF file from their custom version but due to the age of their Cura (5.3 something), when I opened it in 5.7.2, only the model would come through.  

Bringing these definitions in will give users not only the printer's basic settings but the entire start gcode and default slicer settings.

### Affected users and/or printers

All current Mingda FDM machines

GregValiant, 2024-07-03:
Thanks.
There is a process for adding printers through pull requests.  They go through a laborious process of checking to insure there isn't duplication of settings and that sort of thing.
There are currently 12 Mingda printers supported within Cura.  You have duplicated some of them and any conflicts between the current definition files of those printers, and the ones you have provided, need to be fixed.

The problem I see here is that there is a lot of work to be done going through all the files and making sure everything is copacetic.  I don't know how many people are actually on the Cura Team but I know it isn't a lot (4 or 5?) I'm sure the workload of the Cura Team members keeping up with UltiMaker things is heavy enough that doing all that work on definitions for another companies machines (a company that apparently doesn't care to do it themselves) would not be done until sometime in about the spring of 3754 AD.

I would suggest that you just do a pull request just for the definition for your printer.  There is a much greater chance of getting one definition in than 22 at once.

One more thing...bulls are males, cows are females.  I don't know if she would object, but Slashee is 6'5" and I'm not taking any chances.

LilBub, 2024-07-06:
Well I tried the process Slashee suggested on another machine.

Instead after adding the printer, Cura immediately crashes.

[image]

GregValiant, 2024-07-07:
Which machine did you add that caused the crash?

LilBub, 2024-07-07:
The Mingda Magician Max2

GregValiant, 2024-07-07:
There is a difference in the "Mingda_base.def.json" files.
Within the base file you posted up above there is:
**"preferred_material":["mingda_pla-claret","mingda_pla-purple"],**
The square brackets indicate that "preferred_material" is a Python List variable containing two items.
In addition, the line appears to be missing a "space" between the colon and the opening square bracket.  I don't know how important that is in this case.  A punctuation error can keep a file from loading.

The base file included with Cura 5.7.2 is:
**"preferred_material": "generic_pla",**
Without the square brackets - "preferred_material" is a simple "String" variable.  It is also a "String" in FDMPrinter.def.json.

So the line in the Mingda_base.def.json file you posted here appears to be in error.  When the "List" gets to "variantNode.py" it causes a crash because of the illegal operation being requested.  You can Insert an item INTO a list, or you can concatenate one string ONTO another string, but not vice versa.

LilBub, 2024-07-07:
That’s very interesting 
In one PC it works fine
On the second, doing the same process, it throws that error.

GregValiant, 2024-07-08:
Maybe the logic took a different path.
I don't think the "Preferred Material" should be two materials.  So my guess is that the new "Mingda_base" file is the one in error.  It may work with the Mingda Slicer.

If you change that line to:
"preferred_material": "generic_pla", (which is the default)
You can make the material whatever you want as long as it exists, but it needs to be a single material as a string rather than multiple materials in a list.

LilBub, 2024-07-08:
Just fired up the PC where the Cura / Mingda Cura mash up worked.

So on this PC, with the Magician Max2 selected, the default material is Generic PLA, as you point out.

It's very interesting it worked on one install and not the other.  I'll take a peek at the file you mentioned (if I can find it!).

[image]

GregValiant, 2024-07-08:
The load order is "fdmprinter.def.json" then "mingda_base.def.json" and then the specific model of Mingda printer.  Any over-rides are taken in that order.  That is controlled by the "inherits" line in the definition files.
If that Mingda_base "preferred_material" line is overridden within the Magician_Max2 printer def file, that might straighten out the setting (change it from a "list" to a "string").  If it isn't overridden, then the setting gets through as a "list" and you get the error.

LilBub, 2024-07-08:
Okay, I'm not sure I am at your level where I know where to find said file(s) and what to edit, with something like Notepad ++ I assume?

LilBub, 2024-07-08:
Hi

So I changed that one file exactly as you wrote it.

I then got this error when Cura started up

[image]

When I go into Printers and adding a Mingda machine, now I only see this

[image]

Looking at the file I edited, did I good up the format?

[image]

GregValiant, 2024-07-09:
Yes you did.
"preferred_material": "generic_pla"},

You have "preferred material in there twice and you left the square brackets which indicate a python "list".

My preference is that people learn and do but here ya go.

[mingda_base.def.zip](https://github.com/user-attachments/files/16135438/mingda_base.def.zip)

LilBub, 2024-07-09:
Well I am trying!  Thank you for that

I haven't mucked with Python, just Visual Basic back in early 2000s, some C# and enough vsCode to edit Marlin  

That worked!

GregValiant, 2024-07-09:
I'm actually a VB guy myself  I have an app that I wrote to communicate with my printer.  You need a USB data cable.  It doesn't send Gcode files to the printer, but it can control printing via the SD card and send commands to the printer and you can see the responses from the printer.
[Greg's SD Print Tool.zip](https://github.com/user-attachments/files/16135974/Greg.s.SD.Print.Tool.zip)
[image]

LilBub, 2024-07-09:
Looks a lot like Reptier Host back in the day, or the S3D Machine Control although you have many other features.

Most of my programming went into the laptop inside my all aluminum R2-D2.  I use an xbox game pad to remotely trigger R2's beeps.  Essentially each button is for the program to pull a random sound file from said specific folder.  When I got the data from LucasFilm, the sound files essentially all sounded alike...but I did hear a trend.  "Happy", "Chatty" and "Sad" is what I named the folders as those are what R2's beeps/twirps seem to sound like!  I later added some DirectX capability using a mini project to be used as a second monitor in Windows.  With a key press, it would light up the projector and play the MPG of Princess Leia's holographic message.   This, of course, got ruined by many an unruly child thrashing on my droid and knocking the projector out of alignment with the holoprojector on the dome.  (Not everyone is mindful of their children around a $15,000 droid - ask any antique car guy that displays his autos at shows and how many mirrors and hood ornaments get torqued with!)  

Anyways, way off topic but super fun too.  You can always check out my YouTube (Where Nerdy is Cool) for my 3d printing

With regards to Mingda - hilariously, they seem to just continue to say "Just use Prusa Slicer, we have a profile for that."  And I have had an email conversation with them essentially saying "Look, you have the data, just give Team Cura what they need."  But their answer for their broken fork is use another slicer.  It's dumb.

GregValiant, 2024-07-09:
Arrogant idiots.  They aren't the only ones like that either.

So this client comes in and has a complaint about the marionettes he sees in the New York subway stations.  I get tasked (for good money) to build a marionette that "bows" better.  I ended up with a servo-driven puppet and it was pretty cool.  It could dance, but the "bow" at the end required that the feet stay in place so the torso could bend over without the changing CG of the assembly affecting the movement.  I used magnets in the feet and a metal plate to dance on, but the client was unhappy with the result.
I thought he was nuts, but I got paid.

LilBub, 2024-07-09:
Hey you got paid!  
I'll keep dogging Mingda about this.  They are asking when my review for this printer will go on YouTube and I keep pointing out it took two weeks to successfully slice test prints for it.  (And without their help but the Cura Devs!)

GregValiant, 2024-07-09:
I've never seen one and I can review it right now.
"The fans are junk."
"The "factory assembled" portions weren't put together with any eye towards accuracy and they require a complete re-work by the end user.
The firmware is:  non-standard, important commands have not been enabled, and doesn't talk to the LCD correctly.
If it's a bowden printer then the clips that hold the bowden tube in place are not just junk, but are an improper design for their function.

How'd I do?

Here is a fact I found absolutely amazing:
_The failure rate of the plastic extruder pressure arms on Creality printers was 100% within 2 months of their first use.  Some people received brand new printers with cracked pressure arms that required replacement before they could print their first model._
That went on for 2 years.

LilBub, 2024-07-10:
Okay anything I try to slice crashes, as posted in another ticket

Just for the heck of it, I went through Add a Printer and selected the Mingda Magician X2.  I placed a benchie on the bed, made no changes and hit Slice.  Instant error.

[Mingda_X2_CFFFP_3DBenchy.zip](https://github.com/user-attachments/files/16152713/Mingda_X2_CFFFP_3DBenchy.zip)

I'm really curious where the error is and what file?

LilBub, 2024-07-10:
(Quick side note - I asked Mingda to use the Submit New Feature link in here to add the Mingda profiles.  Perhaps in vain, but wanted to let you know, I tried again!)

GregValiant, 2024-07-11:
The X2 definition file shows the platform as:  "platform": "mingda_magician_x+_base.stl"
The "+" sign is an illegal character and there is no file with that name.  It should be "x2".  I wouldn't think the definition file would load with that error in it.  You should not have been able to install that printer.

LilBub, 2024-07-11:
Which json file is that?  I'll check that out when i get home.

GregValiant, 2024-07-11:
It's the one for the Magician Xplus "mingda_magician_xplus.def.json".
Little things like that get noticed and when they do the file won't load, or will cause Cura to crash when it reads the line with the error.  Sometimes things will show up in the log file, but I don't think a "non-loading" definition file trips an error logging event.  Instead you just get that "Reset Everything" warning that has jumped up and bit me hard a couple of times.

LilBub, 2024-07-12:
Hi @GregValiant 

Made that change, sliced a file, same crash

[image]

GregValiant, 2024-07-12:
Right after a crash get the Cura.log file from the configuration folder, zip it and post it.

LilBub, 2024-07-12:
Before we grafted over the Mingda profiles, my Mingda Magician X2 profile worked fine.  That's blowing up now.
Ran that first.  
Switched the Mingda Magician Max2 profile
Sliced file, crash.
Here's the file!
[cura.zip](https://github.com/user-attachments/files/16186532/cura.zip)

LilBub, 2024-07-12:
Hi @GregValiant , checking in.  Any revelations in the log file?  Thanks!

GregValiant, 2024-07-12:
There are loads of errors.  I'm assuming that many are from the mingda definitions but it's hard to tell.
I think you need to delete all the new definition files from Cura, then bring one in at a time and see if it will load and slice.  If it does then it's likely OK.  If it causes a crash then you can send it back to Mingda for a rework.
That also covers the Quality files.  Start with just "Standard" and then move additional ones in and try them.
Any references to the math symbol "+" needs to be removed from all files and replaced with an actual character or number.

LilBub, 2024-07-12:
Oh boy.
So what to files from which folders should go?
Then, let's only put the Max2 files in

LilBub, 2024-07-21:
@HellAholic thanks for looking at this.

This is another one of those issues that explodes on one install, works fine (like yours) on another install.

For what it's worth - the machine is pretty decent for a sub $200 machine.  I'm not shy to call some of these import machines what they are - junk.  But, despite a few design choices I can't figure out - a light by the nozzle that just casts a shadow on your print - makes dialing in that first layer so much fun - it's not terrible.  

I've also made several appeals to Mingda to devote whatever resources they pour into their fork of Cura into working directly with you guys.  Their support team says to go run Prusa Slicer if you have issues with their Cura.   No thanks.

LilBub, 2024-07-22:
Is there anything in particular I should be asking Mingda for?  I don't mind shepherding this along a bit.
These are very inexpensive machines and seem to work well with basic settings.  Where they lack are some advanced things other machines can do with ease (USB printing for Octoprint - their ports are only for media drives)

LilBub, 2024-07-22:
> The main part missing is the test prints to check for the output. I've asked the Cura team to give feedback on the profiles, so few of the issues and things that I've missed will probably get resolved with the review. If we can somehow get some prints going with the modified definitions, that would be good enough to merge for the next Beta period.
> Since the configuration folder supports adding new printers the testing can be done on the either 5.8 Beta or the 5.8 stable.
> Let me know if you can do a few test prints or if the manufacturer is open to testing and updating the profiles going forward.

Sure, I’m willing to help do a few test prints with the Magician X2 and Magician Max2Sent from my iPhone