Ultimaker/Cura #19330: missing field

XelaCerv, 2024-07-06:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 Pro 

### Name abnormal settings

Changing nozzle size from .2mm to .8mm 

### Describe model location

No

### Describe your model

Nothing saves anymore after slicing

### Add your .zip here ⬇️

dont know what a .zip is

GregValiant, 2024-07-06:
Thanks for the report.
I installed  an Elegoo 3 Pro with a 0.8 nozzle.  I didn't have any trouble slicing or saving.
Without more information there isn't anything else that can be done.
Does it save if you switch back to the 0.4 nozzle?

XelaCerv, 2024-07-06:
I switched back to .4 but still will not save

Thank you,
Alex Cervantes

GregValiant, 2024-07-06:
Then it is a local problem with your system.  Cura makes calls to Windows API and Windows actually does the save.
You can try right-clicking on the desktop icon and select "Run as Administrator" and see if it makes a difference.

I'll go ahead and close this as it isn't likely to be a Cura problem.  I hope you understand.

closed 2024-07-06 by GregValiant

XelaCerv, 2024-07-06:
It is a cura problem because this happens on my computer and my friends computer that I logged into aswell

Thank you,
Alex Cervantes

GregValiant, 2024-07-06:
I can't duplicate the problem and there haven't been any other bug reports regarding this.  Tens of thousands of people use Cura and this just hasn't been an issue.  I have 7 different versions loaded and all of them work correctly.
Use the "Help | Show Configuration Folder" command.  In that folder is the Cura.log file.  Zip the log file (ask somebody) and post the zip folder here.  If it shows what the problem is I can re-open this.