Ultimaker/Cura #19331: Definitions for Creality K1 series Printers

obbert42, 2024-07-06:
### Is your feature request related to a problem?

There is no profile for the Creality K1 series printers

### Describe the solution you'd like

implement a profile for Creality K1 series printers

### Describe alternatives you've considered

I currently use competing software

### Affected users and/or printers

Anyone who owns a Creality K1 Speedy, Creality K1 Max, or Creality K1c.

### Additional information & file uploads

You have released profiles for Creality's other newest line of printers (ie Ender 3 v3) yet the k1 line seems to elude you???

GregValiant, 2024-07-06:
The profiles for 3rd party printers must come from the manufacturer, or from a member of the community who owns one.
I see that a pull request ( #17531 ) for the K1C and ( [#17871 )](https://github.com/Ultimaker/Cura/pull/17871) K1MAX have been submitted.  I don't know if they cover the other sub-models.

I'll leave this open for now.

SPlatten, 2024-10-23:
Is there a working profile for the Creality K1 MAX for Cura ?

GregValiant, 2024-10-23:
There are printer definition files for the K1 Max included with (at least) 5.8.0.  It should include the base quality profiles.  Whether they work or not I have no idea.

SPlatten, 2024-10-23:
Perhaps its the MAC version, the printer is visible on my WiFi network however I cannot add the printer using the IP and I click on the Non Ultimaker Printer, I've clicked on the link "Can't connect to your Ultimate printer?", it doesn't give me anything that I could try, where would the definitions be for the K1 Max and how do I enable and select them?

GregValiant, 2024-10-23:
WiFi printing from UM Cura is for UltiMaker printers.  I don't think it works with any third party printers.
You might try slicing with UM Cura and then open the gcode in Creality Cura and send it over the network from there.  I have no idea if that will work but it seems it should.
Another option would be OctoPrint.

GregValiant, 2024-10-24:
_"...the model in the gcode is way off scale"_
Whoa...hold on for a second.
That's a completely different thing and is impossible for the slicer to do.  The toolpaths in the gcode are based on the slice of the model that was sitting on the slicer build plate at the time the "Slice" button was activated.  That scale (from slice to gcode) can only be 1:1.  If you want a different sized print, you must scale the model in the slicer, and then re-slice.

Now if you have some interface between the Gcode file and the printer (Octoprint, or a Sonic Pad might be examples) and if that interface had the capability to re-scale a gcode file on the fly, then a change of scale could be possible.  That's not a slicer issue.
But simply printing the gcode file directly (like from an SD card or through the USB port) the scale will be 1:1.  I suppose it would be possible that the K1 itself has a setting to "scale" the gcode.  There is no single Gcode command for "scaling" and so it would have to be some sort of proprietary function built into the firmware.  I have never heard of such a thing though.

I would suggest simplifying your system to it's most basic configuration.  Just slice the model, save the Gcode to your hard drive (so you have a "virgin" version), copy that file to an SD card, stick the card into the printer, and print the file.
If that works - then add a single item (like the WiFi components) and see how it goes.  You need to find out where the problem is.

SPlatten, 2024-10-24:
In the end I used OrcaSlicer to prepare the model and slice it, then put the gcode on a USB stick and its printing find now, couldn't use "Creality Print"

GregValiant, 2024-10-24:
Dealing with 3rd party printers must be done by owners or the manufacturer.  Creality has it's own fork of Cura and so does not always supply UltiMaker with the needed files (whether they be definitions, nozzles, or quality files).
UM Cura does have definition files for the K1 Max and that's all that can be done from this end.

I'm going to change this to a "discussion" and close it.  Direct support for your Creality printer must come from Creality.

closed 2024-10-24 by GregValiant