Ultimaker/Cura #19332: Single wall cylinder printing inconsistencies

fredericuslaurentii, 2024-07-06:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Massively customised helloBEEprusa with Klipper

### Reproduction steps

I have tried to print the file in the zip at 60 mm/s speed, 110 % extrusion multiplier, no retraction, ecc.
[image]

### Actual results

There are printing inconsistences, in fact, while some cilinders' walls are beign printed some others are completely skipped
[image]

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[HBEEP_39_bit_dremel_tools.zip](https://github.com/user-attachments/files/16116493/HBEEP_39_bit_dremel_tools.zip)

GregValiant, 2024-07-06:
Thanks for the report.
This is not a Cura problem.  Your selection of settings is creating problems with this particular model.

The "Initial Layer Height" is 0.15 and the Layer Height is 0.40.  With the line width at 0.60 it's putting a lot of heat down all at once on those single extrusion walls and it looks like the previous layer is just softening and smooshing down.  You also have the "Outer Wall Inset" at 0.20.  The walls of the pockets aren't really thick enough to take making them .4mm thinner.
I would suggest that you turn off "Group Outer Walls".  On that model it's causing all those little spots of "Inner Wall" to go down first and then the nozzle comes back for the Outer Walls.  With Group Outer Walls turned off you will see a much more efficient toolpath.
If you really need to adjust the ID of the pockets - "Hole Horizontal Expansion" would be a better choice.

Give those things a try.  I'm going to remove the bug label as this appears to be a settings issue.  I'll leave it open for a bit so you can respond.

fredericuslaurentii, 2024-07-11:
I thank you for sharing your insights that seem to be going in the correct direction. 

The results seem to have improved although there are still problems in the side surface of the cylinder that present themselves as under-extrusions. 

Trying another print of a hole with a larger diameter shows a circumference deformation that does not seem to be caused by problems in the adhesion of the filament to the bed, but rather, I believe, at the circumference starting point.
If it makes sense, could there be a setting that adjusts the flow for the deposition of the first point of the circumference?

[image]
[image]
[image]
[image]
[image]

fredericuslaurentii, 2024-07-22:
Small follow up I believe is somehow related to the problem: by trying to print this part, beyond the height of the two hexagons, a sub-extrusion of the single wall begins to emerge.

![photo_2024-07-22_16-03-57](https://github.com/user-attachments/assets/0abc8a7f-2a14-4
[image]
[HBEEP_Visor insert.zip](https://github.com/user-attachments/files/16334792/HBEEP_Visor.insert.zip)
fdd-b2c0-f180b5857ff0)