Ultimaker/Cura #19335: Line Type Preview Colour Scheme Mismatch

gh13, 2024-07-06:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Ender 3 S1 Pro

### Reproduction steps

Load & Slice Model then select the preview tab displaying the model by the identified line type 

### Actual results

The line colours in the Line Type panel don't match the colours of the related items of the sliced model

[image]

### Expected results

The line colours in the Line Type panel should match the colours of the related items of the sliced model

### Add your .zip and screenshots here ⬇️

[Niche Bellows Adapter V3 (Complete) [Eyrone PLA Silk, LW-0.4mm, LH-0.2mm, I-25%, T-E.215C, B.65C].zip](https://github.com/user-attachments/files/16117261/Niche.Bellows.Adapter.V3.Complete.Eyrone.PLA.Silk.LW-0.4mm.LH-0.2mm.I-25.T-E.215C.B.65C.zip)

GregValiant, 2024-07-06:
Thanks for the report.
It looks like a couple of the colors aren't being adjusted for "Dark" mode.  Support-Interface seems to be the same in both light and dark modes as is "Travel".
"Travel" and "Helpers" are special cases because each topic covers more than one line type.
"Travel with Retraction" are thin cyan lines and "Travel with no retraction" are wide blue lines.
The skirt is a "Helper" and maybe should be yellow like the support structure.  Support-Interface is also part of Helpers and seems to be the same color in Light mode as in Dark mode.  It is a different color than Support.
Infill is cyan so that changes.  Outer Walls are red and inner walls are green.

I really don't see a problem here.  Which one(s) are you referring to?

gh13, 2024-07-07:
It was the Helpers/Infill colour that was confusing me. I've been using the Colourblind Assist Dark theme, with the skirt and support interface, being coloured cyan, which the legend suggests are Infill. 
I thought these would have been coloured as Helpers. But realise there is a need to represent the support interface at least as a different colour to the supports themselves. I see that with the Standard Ultimaker Light & Dark themes also show these as cyan although they don't use cyan for any of the other line types

GregValiant, 2024-07-07:
PrusaSlicer uses a lot more colors and Line Types.  I don't think there is a "better" or "worse" here, just "different".
If you have an idea for improving the "Color Blind Assist" why don't you describe it here.  People who aren't color-blind can't really tell what is needed.

Can I remove the bug label from this?  The "look" might be able to be improved, but that is a "Feature Request" rather than a bug report.

Back-in-the-day CRT televisions had manual control over the color settings.  Going over to my uncles house to watch TV was a trip as he was color-blind and he would set the TV up for himself.  For everyone else watching - his setup was pretty bizarre.

gh13, 2024-07-07:
TBH I didn't pick the colour blind theme due to suffering from the condition I just preferred the more vivid colours in the preview.
 
If I was to make a suggestion it would to not use the colour defined for the skirt & support interface as the colour defined for the Infill as is the case with the other themes. It seems a bit counter intuitive that a theme intended to increase the ability for colour-blind users to differentiate between line types would use the same colour for for multiple line types.

No problem removing the bug label as it is more of an enhancement. Maybe if there are any colour-blind users reading this the could chime in with their thoughts & suggestions

GregValiant, 2024-07-07:
We'll call it an "Improvement".  Thanks for the thoughts.