Ultimaker/Cura #19336: Slicing Failed

Wazzy-q, 2024-07-07:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Sovol SV01 PRO

### Name abnormal settings

Changed the print speed to 60. added a raft and supports

### Describe model location

No

### Describe your model

I sliced it on my old PC with Sovol Cura 1.3.0. New PC can't get to slice on any Cura except for Lychee Slicer

### Add your .zip here ⬇️

https://we.tl/t-jHTc2LbQ8Z

GregValiant, 2024-07-07:
Thanks for the report.
I don't see any errors in the model but there are 3.75 million surfaces.  That's a tough slice when each facet is only 0.05mm².
I was able to slice it on my Win 10 laptop with Cura 5.7.2.  The slice took 12 1/2 minutes and it's a 95 hour print.
Use the "Help | Show Configuration Folder" command and locate the "Cura.log" file.  Zip it and post it directly here (don't email it).
The Cura team will take a look.
[image]

Wazzy-q, 2024-07-08:
Thanks. Ya this model is so textured it makes sense the slice hard. Just weird that my old PC that is half as powerful can slice it on an older version of Sovol3D Cura.  My new one with newer programs can’t. Sounds good I will do that

image.png (view on web)