Ultimaker/Cura #19341: v5.7.2 keeps adding temperature commands on start_gcode despite they are already present

Murilolana, 2024-07-08:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Custom printer with klipper

### Reproduction steps

I've been using cura for a long time now. I have the "{material_print_temperature_layer_0}" in my start_gcode, and it always worked as expected. But since I updated from v5.5.0 to v5.7.2, Cura keeps adding "M104 " and "M109" commands to the begining of the gcode. Couldn't find the cause for this, as it always worked fine before. Here is my start_gcode config:

"machine_start_gcode": {
                "default_value": ";Sliced at: {day} {date} {time}\n;Filament weight = {filament_weight}\n;Nozzle diameter = {machine_nozzle_size}\n;Filament type = {material_type}\n;Filament name = {material_name}\nPARAMETERS_MATCH MACHINE='{machine_name}' NZ0='{machine_nozzle_id, 0}' NZ1='{machine_nozzle_id, 1}' MAT0='{material_type, 0}' MAT1='{material_type, 1}' F='{filament_weight}' ENABLED='{extruders_enabled_count}' INITIAL_EXT='{initial_extruder_nr}'\nM605 S{0 if idex_mode == 'idex' else 1 if idex_mode == 'copy' else 2}\nM900 P{machine_nozzle_size} S{speed_print}\nCTEMP MAT='{material_type}' FAM='{material_family}'\nBREAK_CLEAN P={material_break_temperature}\nG21 ; set units to millimeters\nG90 ; use absolute positioning\nM82 ; absolute extrusion mode\nM104 T{initial_extruder_nr} S150 ; set extruder temp\nM140 S{material_bed_temperature_layer_0, extruder_nr} ; set bed temp\nG28 Z ; home z\nM109 S150\nG28 XY; home XY\nSTART_GCODE\nset_temperature_layer L0={material_print_temperature_layer_0} L1={material_print_temperature_layer_0, 1}  ; wait for extruder temp\nPRIME_ROUTINE"
                
            I appreciate some help with this :)

### Actual results

T0
M104 S235
M105
M109 S235
M105
M109 T1 S120
M82 ;absolute extrusion mode
;Sliced at: Mon 08-07-2024 09:31:28
;Filament weight = [0.9321797058105469, 0.04065794128417968]
;Nozzle diameter = 0.4
;Filament type = ABS
;Filament name = Black ABS
PARAMETERS_MATCH MACHINE='Syncraft IDEX' NZ0='Standard 0.4mm' NZ1='Standard 0.4mm' MAT0='ABS' MAT1='PVA' F='[0.9321797058105469, 0.04065794128417968]' ENABLED='2' INITIAL_EXT='0'
M605 S0
M900 P0.4 S70
CTEMP MAT='ABS' FAM='H_CHAMBER_TEMP'
BREAK_CLEAN P=50
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 T0 S150 ; set extruder temp
M140 S90 ; set bed temp
G28 Z ; home z
M109 S150
G28 XY; home XY
START_GCODE
set_temperature_layer L0=235 L1=225  ; wait for extruder temp
PRIME_ROUTINE

### Expected results

T0
M82 ;absolute extrusion mode
;Sliced at: Mon 08-07-2024 09:31:28
;Filament weight = [0.9321797058105469, 0.04065794128417968]
;Nozzle diameter = 0.4
;Filament type = ABS
;Filament name = Black ABS
PARAMETERS_MATCH MACHINE='Syncraft IDEX' NZ0='Standard 0.4mm' NZ1='Standard 0.4mm' MAT0='ABS' MAT1='PVA' F='[0.9321797058105469, 0.04065794128417968]' ENABLED='2' INITIAL_EXT='0'
M605 S0
M900 P0.4 S70
CTEMP MAT='ABS' FAM='H_CHAMBER_TEMP'
BREAK_CLEAN P=50
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 T0 S150 ; set extruder temp
M140 S90 ; set bed temp
G28 Z ; home z
M109 S150
G28 XY; home XY
START_GCODE
set_temperature_layer L0=235 L1=225  ; wait for extruder temp
PRIME_ROUTINE

### Add your .zip and screenshots here ⬇️

[start_gcode test.zip](https://github.com/user-attachments/files/16128366/start_gcode.test.zip)

GregValiant, 2024-07-08:
Thanks for the report.
This is a duplicate of numerous other reports.  I'll mark this one and close it in favor of #19204 .

In Cura - use the "Help | Show Configuration Folder".  Unzip the attached file and put "CuraPrependBugFix.py" into your "Configuration Folder" and the "scripts" sub-folder.  It will be available in Cura under "Extensions | Post Processing | Modify Gcode" and then "Add Script".  The only setting is "Enable".  It will remove those lines for you.
Hopefully this will be fixed in 5.8 because it's driving a lot of people nuts.
[CuraPrependBugFix.zip](https://github.com/user-attachments/files/16132125/CuraPrependBugFix.zip)

[image]

closed 2024-07-08 by GregValiant

Murilolana, 2024-07-08:
Thank you very much Greg.