Ultimaker/Cura #19344: Opening STL immediately after starting Cura results in infinite slicing loop

ElijahCuff, 2024-07-09:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Anycubic Kobra 2

### Describe your model

Cross with hole for Pool Que

### Add your .zip here ⬇️

[Uploading AK2_snookerX.zip…]()

GregValiant, 2024-07-09:
Thanks for the report.
That zip file didn't upload correctly and the link just bounces back to this page.  If you didn't include the log file, please do and then try again.

zhou123aw, 2024-07-11:
Have you solved this problem? I am also in the same situation too

GregValiant, 2024-07-11:
I can't investigate because @ElijahCuff 's uploaded file is glitched.  We have nothing to look at.
We need a Project File "zipped" and please include the Cura.log file.

closed 2024-07-24 by github-actions[bot]