Ultimaker/Cura #19348: vertical printing

SamWiseLouis, 2024-07-09:
### Cura Version

5.4.0

### Operating System

linux

### Printer

Elagoo Neptune 3 plus

### Reproduction steps

took a model part for a star wars interdictor and instead of printing flat turned it so it was vertical pointing so i would use less supports and such. IT may be the angle of the parts but i am unsure

### Actual results

Got an error slicing

### Expected results

it to print quicker with less filament

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-07-09:
Thanks for the report.
If you look closely at the models in the Prepare screen you will see "polka-dot" patterns.  Cura uses that pattern to indicate surfaces that are configured incorrectly.  All three parts need repair.  Even if you got those models to slice when laying down you could not trust the slice.
Cura 5.7.1 is much improved over 5.5 and 5.6 so I suggest you upgrade.  5.7.2 is good but it requires a post processing patch because of a problem with it adding temperatures before the StartUp Gcode on ABL equipped machines.

I'm going to go ahead and remove the bug label and close this as it is a model problem and your Cura version is 6 releases behind.  I hope you understand.

closed 2024-07-09 by GregValiant

SamWiseLouis, 2024-07-09:
Thanks i missed that! Much appreciatedSent from my Verizon, Samsung Galaxy smartphone