Ultimaker/Cura #19359: Cura 5.7.2 auto inserting M104, M105, M109 before start code even though they are defined in start code

cmdrscotty, 2024-07-10:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Creality Ender 3 V2

### Reproduction steps

Installed Cura 5.7.2 (previously using 5.4)
Verified same machine setup and start/end Gcodes between 5.4 and 5.7.2
sliced and sent to printer for processing

note, this did not happen in 5.4 (example files attached)

### Actual results

M104, M105, and M109 are auto inserted before start code (didn't do this in 5.4)
Causes printer to heat extruder first even though start codes macro definitions for bed temp and nozzle temp are already defined in start code.

### Expected results

m104, m105, and m109 commands should not be auto inserted if temperature macros are already defined in start code.

### Add your .zip and screenshots here ⬇️

[Cura issue.zip](https://github.com/user-attachments/files/16169344/Cura.issue.zip)

LilBub, 2024-07-11:
Just trying to be helpful - I wonder if this is similar to this reported prepend temperature issue
https://github.com/Ultimaker/Cura/issues/19204

Asterchades, 2024-07-11:
Exactly the same issue, yes. If you look [further down that same report](https://github.com/Ultimaker/Cura/issues/19204#issuecomment-2163070633), user @GregValiant has also been good enough to provide a post-processing script that works around the issue until such time as a new version is released to fix it more permanently.

closed 2024-07-11 by Asterchades