Ultimaker/Cura #19360: Slim and small object fail

Schnooky, 2024-07-10:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Elegoo Neptune 4 Max

### Describe your model

No, I think it's too small and slim.

### Add your .zip here ⬇️

Here is the stl file...
[Funky Inari C.zip](https://github.com/user-attachments/files/16169841/Funky.Inari.C.zip)

GregValiant, 2024-07-11:
Thanks for the report.
I'm not sure what is going on with that model.  When I open the file in Cura it is 6.603mm tall but when I open it in MS 3D Builder it is 66.03mm tall.  My best guess is that it was designed in "cm" rather than "mm".
How tall is the model supposed to be?  If it is only 6.6mm tall then it's just too small to slice even with a 0.2 nozzle.  You can attempt to use "Remove all Holes" and set the Infill Density to 0%, the tops/bottoms to 0, and the Wall Line Count to 1 and see how it goes.  That would slice like this:
[image]
In that slice you can see that the walls end up about 4 times thicker than the model shows.

I don't think this is a Cura bug and I'll remove the "bug" and "Slicing Error" labels.  Let us know about the size.

Schnooky, 2024-07-14:
Thank you for your reply,

I had designed this part - in millimeters - to repair a half-broken Toslink connector, but it seems that it is indeed too thin to be printed.

I've attached a screenshot from Tinkercad.

 

image.png (view on web) <https://github.com/Ultimaker/Cura/assets/64202104/408ab428-d70c-4e9f-a0f3-31b9a1369b7d> 

Cet e-mail a été vérifié par le logiciel antivirus d'AVG.
www.avg.com

GregValiant, 2024-07-15:
I'll go ahead and close this then.  The walls just aren't thick enough to print.

closed 2024-07-15 by GregValiant