Ultimaker/Cura #19364: Cura M104 in Gcode

FrancoCoetzer, 2024-07-11:
### Cura Version

5.7.2

### Operating System

windows 11 & Linux

### Printer

Creality Ender 5 +

### Reproduction steps

Slicing anything now introduces M104, M105, M109 even if my Start Gcode Includes {material_bed_temperature_layer_0} and {material_print_temperature_layer_0}.

Here is my start gcode:
;Cura Start G-code==============================================================================

;Filament weight = {filament_weight}
;Nozzle diameter = {machine_nozzle_size}
;Filament type = {material_type}
;Filament name = {material_name}

START_PRINT BED_TEMP={material_bed_temperature_layer_0} EXTRUDER_TEMP={material_print_temperature_layer_0}

;Cura Start G-code==============================================================================

### Actual results

; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 5 known objects
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_1 CENTER=169.107,154.423 POLYGON=[[159.331,144.634],[159.331,164.13],[178.829,164.13],[178.829,144.634]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl CENTER=174.961,174.966 POLYGON=[[165.251,165.25],[165.251,184.746],[184.749,184.746],[184.749,165.25]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_2 CENTER=180.907,195.584 POLYGON=[[171.147,185.866],[171.147,205.362],[190.645,205.362],[190.645,185.866]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_4 CENTER=195.646,169.177 POLYGON=[[185.91,159.347],[185.91,178.845],[205.406,178.845],[205.406,159.347]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_3 CENTER=154.634,180.885 POLYGON=[[136.588,136.639],[136.588,213.367],[213.407,213.367],[213.407,136.639]]
M104 S245
M105
M109 S245
M82 ;absolute extrusion mode

;Cura Start G-code==============================================================================

;Filament weight = [9.769091025390624]
;Nozzle diameter = 0.5
;Filament type = TPU 95A
;Filament name = TPU 95A

START_PRINT BED_TEMP=0 EXTRUDER_TEMP=245.0

;Cura Start G-code==============================================================================

### Expected results

; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 5 known objects
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_1 CENTER=169.107,154.423 POLYGON=[[159.331,144.634],[159.331,164.13],[178.829,164.13],[178.829,144.634]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl CENTER=174.961,174.966 POLYGON=[[165.251,165.25],[165.251,184.746],[184.749,184.746],[184.749,165.25]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_2 CENTER=180.907,195.584 POLYGON=[[171.147,185.866],[171.147,205.362],[190.645,205.362],[190.645,185.866]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_4 CENTER=195.646,169.177 POLYGON=[[185.91,159.347],[185.91,178.845],[205.406,178.845],[205.406,159.347]]
EXCLUDE_OBJECT_DEFINE NAME=Rubber_Stand_TPE_stl_3 CENTER=154.634,180.885 POLYGON=[[136.588,136.639],[136.588,213.367],[213.407,213.367],[213.407,136.639]]
M82 ;absolute extrusion mode

;Cura Start G-code==============================================================================

;Filament weight = [9.769091025390624]
;Nozzle diameter = 0.5
;Filament type = TPU 95A
;Filament name = TPU 95A

START_PRINT BED_TEMP=0 EXTRUDER_TEMP=245.0

;Cura Start G-code==============================================================================

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-07-11:
Thanks for the report.
Within report #19204 I posted a zip file containing a post-processor to get rid of the "Prepend Temperature" lines when they aren't needed.  Have you installed and activated "CuraPrependBugFix.py"?
It checks to see if "material_print_temperature" and "material_bed_temperature" are in the StartUp Gcode and if they are - then it deletes the prepend temperatures as needed.
If you haven't installed it then I suggest you try it.  If you have installed it and it isn't working for you please let me know.

This appears to be a duplicate of #19204 and I'll mark it as such.  I'm going to leave this open until you respond.  At that point I'll make a further decision as to whether to close this if it is a true duplicate of the other report and if the post-processor fixes the problem.

MariMakes, 2024-07-16:
👋 Quick update on our side.
This is resolved in the 5.8 release 🎉 , you can download the Beta version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-07-16 by MariMakes