Ultimaker/Cura #19367: Cura Tests all ports/devices before allowing model import

Beynon5, 2024-07-11:
### Cura Version

5.7.0

### Operating System

MacOS Monterey

### Printer

Sunlu S9+ / ANET A8 Plus / Wanhao Duplicator I3

### Reproduction steps

Open Cura - Try to import model either by going file > open or drag drop

### Actual results

Model doesn't appear for around 5 - 10 minutes until the port testing has finished

### Expected results

Model imports while port test happens

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/16178822/cura.log)

Here is the part where its testing ports:
2024-07-10 11:43:56,478 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:43:57,025 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT2 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:43:57,208 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:43:58,431 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2024-07-10 11:43:58,957 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/Applications/UltiMaker Cura.app/Contents/Resources/share/uranium/resources/shaders/default.shader]...
2024-07-10 11:43:58,962 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [/Applications/UltiMaker Cura.app/Contents/Resources/share/cura/resources/shaders/grid.shader]...
2024-07-10 11:44:00,814 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [221]: Error check finished, result = False, time = 13.52s
2024-07-10 11:44:01,004 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:01,712 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT2 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:02,002 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:03,307 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 115200 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,309 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 115200
2024-07-10 11:44:03,310 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 250000 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,311 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 250000
2024-07-10 11:44:03,312 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 500000 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,312 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 500000
2024-07-10 11:44:03,313 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 230400 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,314 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 230400
2024-07-10 11:44:03,315 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 76800 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,318 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 76800
2024-07-10 11:44:03,320 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 57600 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,321 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 57600
2024-07-10 11:44:03,324 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 38400 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,325 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 38400
2024-07-10 11:44:03,325 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 19200 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,326 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 19200
2024-07-10 11:44:03,327 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.BLTH if baud rate 9600 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:44:03,327 - WARNING - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [45]: Unable to create serial connection to None with baud rate 9600
2024-07-10 11:44:04,512 - INFO - [MainThread] cura.CuraApplication.readLocalFile [2006]: Attempting to read file file:///Users/chrisbeynon/Downloads/Tora_print files/3D models/Tora_44.stl
2024-07-10 11:44:05,574 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:06,329 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT2 if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:06,623 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:08,755 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.039549827575683594 seconds
2024-07-10 11:44:10,088 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:10,975 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT2 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:11,264 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:14,595 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:15,612 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT2 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:15,878 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2024-07-10 11:44:18,331 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [77]: Unable to find a working baudrate for None
2024-07-10 11:44:18,342 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [21]: Auto detect baud rate started for /dev/cu.Bluetooth-Incoming-Port
[... 28 lines cut ...]
2024-07-10 11:45:43,248 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 230400 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:45:44,946 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT2 if baud rate 230400 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:45:45,240 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 230400 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:45:54,067 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.Bluetooth-Incoming-Port if baud rate 500000 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:46:03,264 - DEBUG - [JobQueueWorker [3]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.GalaxyBudsPro24A5 if baud rate 76800 works. Retry nr: 1. Wait timeout: 15
[... cut ...]
2024-07-10 11:47:25,929 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.URT1 if baud rate 9600 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:47:34,203 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.Bluetooth-Incoming-Port if baud rate 19200 works. Retry nr: 1. Wait timeout: 15
2024-07-10 11:47:54,232 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [39]: Checking /dev/cu.Bluetooth-Incoming-Port if baud rate 9600 works. Retry nr: 1. Wait timeout: 15

GregValiant, 2024-07-12:
Thanks for the report.
If you aren't using USB printing:  Go to the Cura MarketPlace, select the "gear" icon, scroll down and disable the "USB Printing" plugin.
If one of the devices you have plugged into a USB port is configured as a serial device (looks like GalaxyBudsPro24A5 might be) then USB printing will keep polling it for a response thinking that it's a printer.
Let us know if that works for you.

Beynon5, 2024-07-12:
> Thanks for the report. If you aren't using USB printing: Go to the Cura MarketPlace, select the "gear" icon, scroll down and disable the "USB Printing" plugin. If one of the devices you have plugged [...]

Perfect! Works for me - thank you!

GregValiant, 2024-07-12:
There ya go.
I'll go ahead and close this as it expected behavior.
Currently USB Printing is "enabled" by default.  I'm considering putting in a Pull Request to change that to "disabled" by default.  More devices are being used that emulate serial devices and the problem you experienced is becoming more widespread.

closed 2024-07-12 by GregValiant