Ultimaker/Cura #19369: 5.7.x Does not work with WAYLAND

KLudwig-V, 2024-07-12:
### Cura Version

5.7.0 Alpha onward (including 5.7.2)

### Operating System

Linux Debian Bookworm

### Printer

Not Applicable

### Reproduction steps

1. Starting from "fresh install" (no Cura SW previously running, no config stored on machine)
2. Download any App Image starting with 5.7.0 Alpha
3. Execute the Image

### Actual results

### Downloading 5.7.x and executing DOES NOT WORK

Cura UI does not finish the initialization process.
Blue initial windown is shown. "Load Plugins"..."Initializing Engine" and other messages are shown
Initialization crashes just when executing "Loading UI"
The following info is shown in the console

qt.qpa.wayland: Ignoring unexpected wl_surface.enter received for output with id: 7 screen name: "Screen5" screen model: "unknown" This is most likely a bug in the compositor.
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMar2GNKX/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMar2GNKX/share/cura/resources/qml/Actions.qml"
**The Wayland connection broke. Did the Wayland compositor die?**

And this is the last line written to logs (in all cases) (full logs attached)

DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [38]: No active GlobalStack, set CustomQualityProfilesDropDownMenuModel as empty.

### Downloading 5.6 and executing it WORKS OK

And the following info is shown on the console
qt.qpa.wayland: Ignoring unexpected wl_surface.enter received for output with id: 7 screen name: "Screen5" screen model: "unknown" This is most likely a bug in the compositor.
Cyclic dependency detected between "file:///tmp/.mount_UltiMaZVX55V/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaZVX55V/share/cura/resources/qml/Actions.qml"
qt.qpa.wayland: Ignoring unexpected wl_surface.enter received for output with id: 7 screen name: "Screen5" screen model: "unknown" This is most likely a bug in the compositor.

So it seems the only difference is 
**The Wayland connection broke. Did the Wayland compositor die?**

### Forcing X11 in 5.7.x WORKS OK

_3D-Printer$_ QT_QPA_PLATFORM=xcb ./UltiMaker-Cura-5.7.0-alpha.0-linux-X64.AppImage

Console:
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMad0SHYE/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMad0SHYE/share/cura/resources/qml/Actions.qml"

So clearly the problem is only in 5.7 and with Wayland. And the "Cyclic Dependency detected..." is not part of the issue

### Expected results

It should work under Wayland

### Add your .zip and screenshots here ⬇️

LOGS

[Cura-5.7.x-Wayland_Issue.txt](https://github.com/user-attachments/files/16188284/Cura-5.7.x-Wayland_Issue.txt)

KLudwig-V, 2024-07-12:
By the way, this has been reported in several Issues. But those either were pointing to the wrong root cause, lacked logs, were closed because the reporter did not answer follow-ups... so I decided to open a new & clean one.

This is also reported in several places on the web. e.g.: https://www.reddit.com/r/chromeos/comments/1c8uhai/hi_trying_to_run_ultimaker_cura_in_linux_and_i/

KLudwig-V, 2024-08-31:
No reaction? Isn't anybody triaging bugs?

smartavionics, 2024-08-31:
Hi, I can't help you with 5.7.0 but my Cura 4 variant does appear to run under Wayland (tested with weston). You can find my builds at https://github.com/smartavionics/Cura/releases.

I used QT_QPA_PLATFORM=wayland-egl ./Cura-4.20.23-mb-x86_64.AppImage

KLudwig-V, 2024-09-03:
Yes, anything before 5.7-Alpha works. Anything after that does not

rephlex, 2026-10-25:
Independent of the Wayland question: on bookworm you can drop the AppImage completely. My company maintains a signed APT repo with Cura 5.13.0 built natively for Debian 12:

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/debian/ plustech-bookworm main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

Whether the Wayland rendering issue persists with a native build I cannot promise, but it removes the bundled Qt from the equation and you are on a current version instead of 5.7. https://mirror.plustech.de/ (community repo, sources included).